Ysurac/openmptcprouter #1462: DHCP Bail Infini

JeedHome44, 2020-12-18:
Bonjour,

J'ai cherché dans le wiki et des les issue mais aucune ne m'apporte la réponse.

Je vois que dans le DHCP, on peut choisir la durée du bail indépendamment pour chaque appareil.
Je souhaite mettre des baux infinis pour mes appareils fixés.

Il n'y a pas d'exemple sous la case "Durée du bail" pour savoir comment spécifier de le mettre à l'infini.
Faut-il l'écrire en anglais ou en français?

Merci

Ysurac, 2020-12-18:
Il n'est pas possible avec le protocole DHCP d'avoir un bail infini.

JeedHome44, 2020-12-18:
Sur les routeurs TP Link ou autre, on peut voir "Permanent" sur le DHCP. 
Ça signifie quoi dans ce cas?
Mon but étant de laisser les baux pour les appareils fixés.
C'est donc impossible?

Ysurac, 2020-12-18:
Ça veut dire qu'ils mentent. On peut mettre une valeur importante (jusqu'à 135 ans environ) mais pas infinie.

Ysurac, 2020-12-18:
dnsmasq permet de mettre la valeur "infinite" visiblement. Ca fonctionne peut être et je suppose que ça doit mettre la valeur maximale que DHCP autorise (donc autour de 135 ans).

JeedHome44, 2020-12-18:
Déjà 135 ans pour nous c'est l'infini! Je pense pas qu'on soit encore de ce monde dans 135 ans 😉
On peut donc mettre 10 ans?

Comment faut il l'écrire? 10a?

JeedHome44, 2020-12-18:
Ah donc si je mets "infinite" ça doit marcher c'est ça?

Ysurac, 2020-12-18:
oui, visiblement ça supporte en seconde, en minutes (45m), en heures (12h) et "infinite".
Donc en mettant infinite ça doit marcher, je vais ajouter une petite aide car effectivement sans regarder la doc de dnsmasq on a aucune idée de quoi mettre.

JeedHome44, 2020-12-18:
Ok merci 😉
Je vais tester ça!

JeedHome44, 2020-12-18:
j'ai tenté de mettre infinite sur un appareil débranché. je l'ai ensuite branché et ça affiche quand même le réglage de l'interface et pas infinite.

Kalimeiro, 2020-12-18:
Bonjour à tous,

Je confirme qu'en mettant "infinite" cela fonctionne parfaitement, je fais cela depuis plusieurs semaines et aucun problème à constater, si ce n'est les clients DHCP qui de temps en temps renégocie leur IP et je ne pense pas que ce soit le serveur DHCP qui fasse celà.

clients Windows : https://i.imgur.com/MNfnA11.png
omr rpi4 client iphone : https://i.imgur.com/WlX8823.png
omr rpi4 client Windows : https://i.imgur.com/sZHaKgN.png

closed 2021-03-23 by github-actions[bot]