Ysurac/openmptcprouter #1463: TuneIn et Google Home

JeedHome44, 2020-12-18:
Bonjour,

J'ai demandé à un de mes Google Home une radio ce matin.

Il me dit ok puis j'entends une voix en anglais dire que cette radio n'est pas autorisé dans mon pays.

Je demande pourtant FunRadio.

OMR peut il être la cause de ça si on a activé le cryptage avec Shadowsocks?

On n'est plus localisé en France?

Une solution avec un by-pass? Mais comment?

Merci

Ysurac, 2020-12-18:
Possible que l'IP de sortie, donc celle du VPS, soit bloquée. Rien à voir avec OMR directement, juste parfois la décision de certains sites de bloquer des IPs.
Possible avec omr-bypass, il faut cependant trouver le domaine ou IPs utilisées.

JeedHome44, 2020-12-18:
Merci pour la réponse.
Tu n'utilises pas TuneIn avec Google home ou Google assistant?
Ou simplement TuneIn sur smartphone pour voir quels domains bypasser?
Dommage que ça bloque...
Donc un français qui veut écouter des radios FR à l'étranger ne peut donc pas

Ysurac, 2020-12-18:
Non j'utilise rien de tout ça et j'avoue que j'avais jamais entendu parler de TuneIn avant ;)
Tu peux voir sur funradio.fr si jamais tu as le même soucis.
Il y a, comme pour le cinéma, souvent des questions de droits. Dans certains cas c'est aussi de la protection idiote, car je pense que l'IP du VPS est en France.

JeedHome44, 2020-12-18:
J'ai activé le cryptage shadowsocks et ça fait ça.
Oui VPS OVH en France.
Je me demande donc si ce n'est pas le cryptage shadowsocks qui fait ce blocage.
C'est pas possible c'est ça?

Et avec Google Home c'est TuneIn obligatoirement donc pas top!

Ysurac, 2020-12-18:
Le chiffrement de Shadowsocks est activé par défaut et ça ne peut pas influer sur ça.
Tu peux tester sur tunein.com si tu as le même soucis.
Tu peux essayer dans omr-bypass d'ajouter "icecast.funradio.fr"

JeedHome44, 2020-12-18:
Sur l'app TuneIn iOS, j'ai l'impression qu'il me considère aux Royaume unis.
Mon VPS OVH est à Roubaix.
Possible qu'il soit localisé la bas?

JeedHome44, 2020-12-18:
Je pense que c'est TuneIn au complet qu'il faut byPass. Je leur ai envoyé un message pour savoir quoi bypasser

Ysurac, 2020-12-18:
Tu peux ajouter tunein.com en domaine à bypasser alors.
Les bases de données de localisation des IPs sont parfois pas à jour.

JeedHome44, 2020-12-18:
C'est bon ca marche merci 👍

closed 2020-12-18 by JeedHome44

JeedHome44, 2020-12-20:
Bonjour,

Je me permet de réouvrir l'issue car je viens de comprendre.

Si j'active le brouillage Shadowsocks, quand je fais un test NPerf de débit, ça me situe à Londres.
Et si je désactive le brouillage, je suis bien en France.

Mais, j'ai toujours le problème de wan2 instable avec la version 56. Le problème disparait si j'active le brouillage Shadowsocks.

Je voulais vous faire ce retour car le gros problème que j'ai eu (et que j'ai encore) avec la version 56 est résolu en activant le brouillage. Brouillage que je ne souhaite pas dans l'idéal pour bien me situer en France et ne pas devoir faire de bypass...