Ysurac/openmptcprouter #1464: total bandwith limited when LTE 4G plan is over (speed limited by Orange Telecom)

lyr847, 2020-12-18:
## Expected Behavior

When my 4G plan is over (100 Gb consumed), speed is limited by Orange, I should see my aggregated OpenMptcp bandwith falling to 12 Mbit/s (my VDSL Box speed).

## Current Behavior

The problem is that my aggregated OpenMptcp bandwith falls to 1-2 Mbit/s instead of 12 Mbit/s (my VDSL speed), when my 4G plan is over (100 Gb consumed).
## Possible Solution

Exclude wan2 (in my case) unless a minimum speed of (12 Mbit/s or greater) is retrieved, to avoid a bottleneck on bandwith caused by LTE limited speed.
## Steps to Reproduce the Problem

  1.Wait for the end of 4G plan. With an unlimited 4G plan, the problem will never appear ! But maybe my problem could appear in another contexts when LTE speed is limited or KO (4G antenna failure, 4G signal degradation,...).

  2.Test speed before and after bandwith fall
  3.See OMR system log

## Context (Environment)

At home, users cry when Internet speed is crappy.

Solution : exclude an interface from agregation if there is a bottleneck on it,  or let it  in a special mode (idle mode with no impact on agregation) unless a minimal speed is retrieved on this interface.

## Specifications

  - OpenMPTCProuter version: 
   v0.56.5 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: 
  0.1022 5.4.74.mptcp
  - OpenMPTCProuter platform: 
    RPI4x86_64

Kalimeiro, 2020-12-18:
What's your master interface ? VDSL BOX or 4G BOX ?

lyr847, 2020-12-19:
VDSL Box is multiptah master (wan1)

Ysurac, 2020-12-19:
Maybe they do something that end/modify TCP packets badly when they limit speed.
I would need the result of Network->MPTCP, multipath support check on the interface when it's not limited and when there is the limit.

lyr847, 2020-12-19:
Ok I can test for these 2 cases, because I have many 4G plans and SIM cards.
Here are check results:

Not Limited MPTCP wan2 test result
_______________________________________
1: 192.168.11.1 
2: 10.68.195.64  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: 10.68.194.62  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
4: 10.68.194.65  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: * 
6: * 
7: * 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: * 
29: * 
30: * 
31: * 
32: * 
33: * 
34: * 
35: * 
36: * 
37: * 
38: * 
39: * 
40: * 
41: * 
42: * 
43: * 
44: * 
45: * 
46: * 
47: * 
48: * 
49: * 
50: * 
51: * 
52: * 
53: * 
54: * 
55: * 
56: * 
57: * 
58: * 
59: * 
60: * 
61: * 
62: * 
63: * 
64: * 

Result:
nil

Limited MPTCP wan2 test result
_______________________________________
1: 192.168.11.1 
2: 10.216.11.64  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: 10.216.10.62  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
4: 10.216.10.65  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 81.253.184.105  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
6: 193.251.110.185  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
7: 193.252.101.234  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
8: * 
9: * 
10: * 
11: * 
12: * 
13: * 
14: * 
15: * 
16: * 
17: * 
18: * 
19: * 
20: * 
21: * 
22: * 
23: * 
24: * 
25: * 
26: * 
27: * 
28: * 
29: * 
30: * 
31: * 
32: * 
33: * 
34: * 
35: * 
36: * 
37: * 
38: * 
39: * 
40: * 
41: * 
42: * 
43: * 
44: * 
45: * 
46: * 
47: * 
48: * 
49: * 
50: * 
51: * 
52: * 
53: * 
54: * 
55: * 
56: * 
57: * 
58: * 
59: * 
60: * 
61: * 
62: * 
63: * 
64: * 

Result:
nil

No difference, no visible problem !

lyr847, 2020-12-20:
How do ISPs limit bandwith, if they do not end/modify TCP badly ? By increasing latency or gigue ? 
Anyway,  Does Multipath-TCP have a mechanism that prevents a bad connection/interface speed, from impacting the total bandwidth ?

Ysurac, 2020-12-20:
By limiting only bandwidth without doing anything with packets.
If packets are modified it's possible that Multipath tag in TCP optional header are removed and multipath doesn't work at all.
In this case the only way I see would be to do a script that detect limited bandwitdh and add/remove interface.

lyr847, 2020-12-28:
Is is not a better idea to switch off  interface and delete route rather than delete interface ?  When I switch off the LTE router,I see in system log that wan is automatically switched off and route is deleted by OMR. When I switch on, all is automatically remount.

Ysurac, 2020-12-29:
Yes, when I was speaking about add/remove interface, it's add/remove interface in multipath. It's also possible to up/down interface via ifdown/ifup.

lyr847, 2020-12-29:
Ok great. Do you think to integrate this functionnality to a next OpenMPTCProuter version ? Activated or not by a button, selection of interface to be monitored, and trigger threshold speed value to define ?

Ysurac, 2020-12-29:
I don't know any easy way to detect when ISP is throttling/limiting speed.

lyr847, 2020-12-29:
Effectively, but if you real-time monitor average bandwith on a choosen interface and you define a trigger threshold speed value  (e.g : 2 Mbits/s (in fact when ISP limits bandwith). For a 4G connexion it is anormaly very slow, so the script will switch off wan interface, as long as the speed 2 Mbits/s is not exceeded  (if ISP no longer limits access, the script will switch on wan interface) . This is just an idea.

closed 2021-04-04 by github-actions[bot]