Ysurac/openmptcprouter #1465: RPI4 - Crash on boot if no HDMI plugged-in

ArChlOgY, 2020-12-19:
Good day,

I discovered a strange behavior from my RPI4 and the last OMR version v0.56.*, also beta v0.57.*. 
I think this issue started after v0.54.7 since I had no issue with this one.

So I did a fresh install from scratch by following the step by step guide on website. Put the SD card into the RPI then start it.
I had not answer on local network to reach the interface, the network leds on RPI were stuck as well.

I finally connected the RPI4 on my TV to check the boot process and see that it works well ! I plugged the network as well and I am now able to reach the interface. 
I did basic configuration and everything is just fine.

Then I remove the HDMI cable from TV, try to boot the RPI4 and got the issue again ! 

In conclusion I think the RPI & OMR image request HDMI plug-in , if not it just crash or shutdown itself.

Ysurac, 2020-12-19:
I use a RPI4 as main router, so I can say it's not the case.

Ysurac, 2020-12-19:
You should check your power supply, as CPU is used at max frequency this use more power than raspbian by default.

ArChlOgY, 2020-12-20:
You are correct Ysurac ! Another power supply and it runs like a charm.

I can't figure out why I plugged HDMI the RPI4 started 🤔

closed 2020-12-20 by ArChlOgY