Ysurac/openmptcprouter #1467: Erreur maj VPS

Nadeistos, 2020-12-22:
Bonjour,

MAJ OpenMPTCProuter VPS 0.1022

J'ai cette erreur pour la maj.... :/
a la premiere tentative tout allait bien jusqu'a ca :

(Reading database ... 96453 files and directories currently installed.)
Preparing to unpack /tmp/v2ray-4.31.0-amd64.deb ...
Failed to stop v2ray.service: Unit v2ray.service not loaded.
Failed to disable unit: Unit file v2ray.service does not exist.
dpkg: warning: old v2ray package pre-removal script subprocess returned error exit status 1
dpkg: trying script from the new package instead ...
unlink: cannot unlink '/etc/v2ray/config.json': No such file or directory
Failed to stop v2ray.service: Unit v2ray.service not loaded.
Failed to disable unit: Unit file v2ray.service does not exist.
dpkg: error processing archive /tmp/v2ray-4.31.0-amd64.deb (--install):
 new v2ray package pre-removal script subprocess returned error exit status 1
port: 15437
id: 540d5cb4-5e57-42a8-81c5-f1e00bc0db58
Failed to enable unit: Unit file v2ray.service does not exist.
Failed to start v2ray.service: Unit v2ray.service not found.
dpkg: error while cleaning up:
 installed v2ray package post-installation script subprocess returned error exit status 5
Errors were encountered while processing:
 /tmp/v2ray-4.31.0-amd64.deb

Ensuite lorsque je relance je tombe direct sur ceci :
Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Remove lock and update packages list...
Hit:1 http://security.debian.org/debian-security buster/updates InRelease
Hit:2 http://httpredir.debian.org/debian buster InRelease
Hit:3 http://httpredir.debian.org/debian buster-updates InRelease
Hit:4 https://repo.openmptcprouter.com stretch InRelease
Reading package lists... Done
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree
Reading state information... Done
gnupg is already the newest version (2.2.12-1+deb10u1).
openssh-server is already the newest version (1:7.9p1-10+deb10u2).
apt-transport-https is already the newest version (1.8.2.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/8570 kB of archives.
After this operation, 0 B of additional disk space will be used.
dpkg: error processing package v2ray (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Errors were encountered while processing:
 v2ray
E: Sub-process /usr/bin/dpkg returned an error code (1)

Une idée ? ^^"

Ysurac, 2020-12-22:
Je viens de corriger.
Pour corriger le soucis, un ```apt-get clean``` et éventuellement un ```apt-get remove v2ray```.

Nadeistos, 2020-12-22:
Toujours une erreur...
dpkg: erreur de traitement du paquet v2ray (--remove) :
 Le paquet est dans un état incohérent; vous devriez
 le réinstaller avant d'essayer de le supprimer.
Des erreurs ont été rencontrées pendant l'exécution :
 v2ray
E: Sub-process /usr/bin/dpkg returned an error code (1)

et avec apt-get reinstall v2ray
Préparation du dépaquetage de .../v2ray_4.28.2_amd64.deb ...
Failed to stop v2ray.service: Unit v2ray.service not loaded.
Failed to disable unit: Unit file v2ray.service does not exist.
dpkg: avertissement: old v2ray package pre-removal script subprocess returned error exit status 1
dpkg: tentative d'exécution du script du nouveau paquet à la place...
unlink: impossible de supprimer '/etc/v2ray/config.json': Aucun fichier ou dossier de ce type
Failed to stop v2ray.service: Unit v2ray.service not loaded.
Failed to disable unit: Unit file v2ray.service does not exist.
dpkg: erreur de traitement de l'archive /var/cache/apt/archives/v2ray_4.28.2_amd64.deb (--unpack) :
 new v2ray package pre-removal script subprocess returned error exit status 1
port: 12610
id: 412cccbd-c3d1-4897-8730-e4c41739393c
Failed to enable unit: Unit file v2ray.service does not exist.
Failed to start v2ray.service: Unit v2ray.service not found.
dpkg: error while cleaning up:
 installed v2ray package post-installation script subprocess returned error exit status 5
Des erreurs ont été rencontrées pendant l'exécution :
 /var/cache/apt/archives/v2ray_4.28.2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Ysurac, 2020-12-22:
```dpkg --remove --force-remove-reinstreq v2ray```
Par contre il ne faut pas faire un apt-get reinstall ou essayer de le réinstaller manuellement.

Nadeistos, 2020-12-22:
dpkg: avertissement: problème contourné par utilisation de --force :
dpkg: avertissement: Le paquet est dans un état incohérent; vous devriez
 le réinstaller avant d'essayer de le supprimer.
(Lecture de la base de données... 96453 fichiers et répertoires déjà installés.)
Suppression de v2ray (4.28.2) ...
Failed to stop v2ray.service: Unit v2ray.service not loaded.
Failed to disable unit: Unit file v2ray.service does not exist.
dpkg: erreur de traitement du paquet v2ray (--remove) :
 installed v2ray package pre-removal script subprocess returned error exit status 1
port: 22538
id: 36e3bba8-3253-44e2-8b2d-68fd6ff6b2b2
Failed to enable unit: Unit file v2ray.service does not exist.
Failed to start v2ray.service: Unit v2ray.service not found.
dpkg: error while cleaning up:
 installed v2ray package post-installation script subprocess returned error exit status 5
Des erreurs ont été rencontrées pendant l'exécution :
 v2ray

Il est coriace... :'(
Bon si le problème ne vient que de moi je réinstalle le vps c'est pas un soucis, le probleme est sur 2 vps differents

Ysurac, 2020-12-22:
J'ai corrigé le soucis, mais oui plus simple de réinstaller ;)

Nadeistos, 2020-12-22:
Bon il veut toujours pas lol je reinstall et j'en garde un malade si tu as besoin d'un cobaye au cas ou ;)

Malaga82, 2020-12-22:
i had the same problem, manually created v2ray.service with touch `v2ray.service` on `/etc/systemd/system`
then deleted v2ray with the same command posted here.
then reinstalled via script, it was stuck again, because the service is empty, nevermind.
it created a fine service named `v2ray.service.dpkg-dist` , deleted `v2ray.service` and renamed `v2ray.service.dpkg-dist` in `v2ray.service`.
Runned again the whole script, done.

Ysurac, 2020-12-22:
Yes, this was the error: I was deleting service file too early.
This is fixed in current script (and I also make script fix previous error).

closed 2020-12-30 by Ysurac