Ysurac/openmptcprouter #1468: Update from 0.56.5 to 0.57

mkernel, 2020-12-22:
## Expected Behavior

I made a configuration backup of my running instance (version 0.56.5). 
Afterwards I setup a fresh install of the new Version 0.57.
I expected to import the configuration backup to get everything set up again.

## Current Behavior

However I got the following error when accessing the web interface:
`/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'`

I inspected the File /etc/config/luci on 0.56.5 and on 0.57 before and after importing the configuration backup. I didn't find anything obvious.

## Possible Solution

## Steps to Reproduce the Problem

I'm not sure if it will always fail or if its configuration specific.

  1.  Create a backup from a working openmptcprouter 0.56.5 instance
  2.  Import the backup on a fresh 0.57 instance
  3.  Access the 0.57 instance's web UI.

## Context (Environment)

I'm relying on openmptcprouter for my home internet connection and try to keep it up to date.
I always have a spare environment to test before applying updates, so this is not really affecting me
despite not being able to update.

## Specifications

  - OpenMPTCProuter version: 0.56.5 to 0.57
  - OpenMPTCProuter VPS version: 5.4.74-mptcp 0.1022
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-12-22:
It's often related to a problem in one config file, event if it's not /etc/config/luci
You can check using ```uci show``` via SSH that you get all config.

mkernel, 2020-12-22:
`uci show` dumps out ... a lot of stuff. However a part about main seems to be there and just fine:
```
luci.main=core
luci.main.lang='auto'
luci.main.resourcebase='/luci-static/resources'
luci.main.ubuspath='/ubus'
luci.main.mediaurlbase='/luci-static/openmptcprouter'
```
Should i provide the complete configuration dump on this issue?

Ysurac, 2020-12-22:
in fact the result of ```ls /etc/config/``` and the result of ```uci show``` without keys/server ip.

mkernel, 2020-12-22:
Sure! The contents of /etc/config according to `ls`:
```
dhcp
dropbear
dscp
dsvpn
etherwake
firewall
fstab
glorytun
glorytun-opkg
glorytun-udp
glorytun_recipes
https-dns-proxy
igmpproxy
iperf
luci
luci-opkg
macvlan
mail
mlvpn
network
nginx-ha
nlbwmon
omr-bypass
omr-quota
omr-tracker
openmptcprouter
openvpn
openvpn_recipes
packet_capture
rpcd
shadowsocks-libev
snmpd
sqm
system
ttyd
ucitrack
ucitrack-opkg
uhttpd
unbound
upnpd
v2ray
vnstat
wireless
```
and the uci dump:
[configdump.txt](https://github.com/Ysurac/openmptcprouter/files/5729662/configdump.txt)

Ysurac, 2020-12-22:
In ```logread``` do you have any error about lua or rpcd ?
What is the result if you do a ```/etc/init.d/rpcd restart``` ?
Did you try to reset browser cache ?

mkernel, 2020-12-22:
I have this entry in logread:
`Tue Dec 22 12:26:27 2020 daemon.info procd: Instance rpcd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash`
`/etc/init.d/rpcd restart` doesn't change the situation, except i'm getting the log message again.

running `/sbin/rpcd` over ssh manually allows me to use the Web UI.
Clearing the browser cache doesn't affect the error as well.

Ysurac, 2020-12-22:
Maybe old file that make problem ```rm /var/run/ubus.sock```
Else did you try a reboot ?

mkernel, 2020-12-22:
Yes, i tried rebooting it - without luck.
That file is not present (neither in my configuration backup nor on the machine running 0.57)

Ysurac, 2020-12-22:
Strange that this work when run manually, as init script only do ```rpcd -s /var/run/ubus.sock -t 120```.

mkernel, 2020-12-22:
Ah, i ran it on my first test without any parameters. When executing `rpcd -s /var/run/ubus.sock -t 120`, i get the following error message: `Failed to connect to ubus`

mkernel, 2020-12-22:
Ah! i got something: on 0.56.5 the socket file is at /var/run/ubus.sock, on 0.57 it's at /var/run/ubus/ubus.sock.
I'll change my config and see if that helps after a reboot.

mkernel, 2020-12-22:
That was in deed the problem! I could fix it:
```
uci -q set rpcd.@rpcd[0].socket=/var/run/ubus/ubus.sock
uci -q commit
```

Ysurac, 2020-12-22:
Strange, I didn't have this problem when I updated my old install.
I will add a check for next release.
Thanks :)

mkernel, 2020-12-22:
Thank you for helping me finding it in the first place :)

jeandelamarquise, 2020-12-24:
> That was in deed the problem! I could fix it:
> 
> ```
> uci -q set rpcd.@rpcd[0].socket=/var/run/ubus/ubus.sock
> uci -q commit
> ```

Thanks for your help, i had the same problem ! The 2 commands worked for me too ...
(the rm /var/run/ubus.sock command return no file too. Maybe because it is not at this place like you. I'm not good in linux so I don't try to find it)

Ysurac, 2020-12-24:
This should be fixed in 0.57.1

mkernel, 2020-12-24:
I just retried the update from 0.56 to 0.57.1 (skipping 0.57) and the UI didn't come back up after the configuration restore again. I checked the value of the socket variable and it is wrong again:

```
root@OpenMPTCProuter:~#uci get rpcd.@rpcd[0].socket
/var/run/ubus.sock
```
While the Socket is again located here: `/var/run/ubus/ubus.sock`

xillibit, 2020-12-31:
Need to try if it fixed in 0.57.2

JeedHome44, 2021-01-01:
Problem still there in 0.57.2. But I don't understand how you go about solving it? thank you

mkernel, 2021-01-02:
I redid the update as well, from 0.56.5 to 0.57.2 - with same result as @JeedHome44 and @xillibit - the problem still seems to survive an "rebuild and import backup" update.

mkernel, 2021-01-30:
I just did an update from 0.56.5 to 0.57.3 and this time the UI came back just as expected. So it's fixed, thank you!
I'm going to close the issue now.

closed 2021-01-30 by mkernel

gsxryan, 2021-08-20:
I had this problem on 0.58.5.  It was immediately after looking at realtime graphs > connections.  I cannot reproduce the error now though after having run the workaround above.