Ysurac/openmptcprouter #1470: Update 0.56.5 to 0.57 - Strange logs

Corrado56, 2020-12-22:
## Expected Behavior
No strange logs

## Current Behavior
After the update the OMR router starts correctly, but after few minutes I have continually those errors on System LOG
[image]

And those on Kernel LOG
[image]

BUT except for this strange log, the router is actually working well, and I am not blocked.

## Possible Solution
## Steps to Reproduce the Problem
## Context (Environment)
## Specifications

  - OpenMPTCProuter version:  0.57 
  - OpenMPTCProuter VPS version: 0.1023 
  - OpenMPTCProuter platform: Linksys3200ACM
  
Other info:
  - Sysupdate, not factory install + load configuration backup
  - Rebooted OMR manually after the update

Ysurac, 2020-12-22:
It's related to wifi.

Ysurac, 2020-12-22:
Changing some settings in wifi like "maximum transmit power" and "country" should remove the logs. I will check if I can reproduce and solve the issue (but maybe not before next year).

Corrado56, 2020-12-23:
I tried to change some parameters without success, and the wireless is no more functioning.
those system log lines have some meaning to You ?
[image]

Corrado56, 2020-12-23:
After a reboot wireless restarted and I have no more strange messages.
Ceased alarm

closed 2021-03-28 by github-actions[bot]