Ysurac/openmptcprouter #1473: The ping is unstable since the new version 0.57.1.

cayenne17, 2020-12-23:
## Expected Behavior

Since I upgraded from version 0.56.5 to version 0.57.1, my ping is higher and more unstable. 
I went from 15 ms stable to 30-40ms

## Current Behavior

That the ping is stable

## Possible Solution
When I disconnect the 4g gateways and leave only the ftth connected, the ping is very stable as before (15 ms).

## Steps to Reproduce the Problem

  1. have 2 gateways (one FTTH or XDSL and the other 4g)
  2. be configured a UDP glorytun
  3. FTTH or XDSL on master and 4g on backup
  4. a testé la latence et la gigue avec le client et le serveur teamspeak
  5. problem appears
 

## Context (Environment)
I have 3 WAN connection (2xLTE and 1xFTTH)
My FTTH is master and the 2 LTE is backup
I use glorytun UDP

## Specifications

  - Older OpenMPTCProuter version: 0.56.5
  - Older OpenMPTCProuter VPS version: 0.1022
  - New OpenMPTCProuter version: 0.57.1
  - New OpenMPTCProuter VPS version: 0.1023 5.4.81-mptcp
  - OpenMPTCProuter platform: Virtual machine x86_x64

Configuration before upgrade (0.56.5) :
https://pastebin.com/DQejg4qY

Configuration after update (0.57.1) : 
https://pastebin.com/Ns98r8a2

(0.56.5 and ftth master and 2x4G backup) or (0.57.1 FTTH only) : 
[image]

[image]

(0.57.1 and FTTH master and 2x4G backup)   : 
[image]

[image]

Ysurac, 2020-12-24:
I think I made an error in glorytun udp config, now there is not a backup parameter but some preferences.
I'm compiling a new snapshot with some change that may solve or not the issue, but I don't think I will have time to test until a few days ;)

cayenne17, 2020-12-24:
OK, I'll test when I have time and when the latest version is released.

cayenne17, 2020-12-30:
The problem is solved in commit : https://github.com/Ysurac/openmptcprouter-feeds/commit/60343492d6bf3d1a376efaa6cf2eff5426bef46b#diff-785bdc1682ee35fab7aba0ac039f270bd2a55cb7c09118cc710c4be282ec456b

closed 2020-12-30 by cayenne17