Ysurac/openmptcprouter #1474: Should i update ?

rifaterdemsahin, 2020-12-24:
knock on the wood (: it is working great the last 40 days

[image]

What do I get with the new update and should I also update the VPS as well ?

rifaterdemsahin, 2020-12-24:
Where is the release notes?
not these?
https://github.com/Ysurac/openmptcprouter/commit/1222a62634a33b99caa2780701bddd4416c8ebcd

Maybe these
https://github.com/Ysurac/openmptcprouter/releases

Only git commit are here are there easy to read notes?
https://github.com/Ysurac/openmptcprouter/commit/0991afaa30505f48e55812f61545c9c563e56308

Ysurac, 2020-12-24:
Release notes in on https://www.openmptcprouter.com/ as always.
Detailed changes are on https://github.com/Ysurac/openmptcprouter/commits/master and https://github.com/Ysurac/openmptcprouter-feeds/commits/master

rifaterdemsahin, 2020-12-26:
my bad I should have scrolled
[image]

idea: 
a dedicated page would be easier for us
[image]

closed 2021-04-01 by github-actions[bot]