Ysurac/openmptcprouter #1476: Lost connection VPN glorytun UDP every hour on 0.56.5 and 0.57.1

cayenne17, 2020-12-24:
## Expected Behavior

## Current Behavior
Since I upgraded from version 0.55.3 to version 0.56.5, every hour I have a loss of connection one level of UDP.
Glorytun UDP down.

on version 0.57.1, the bug is still present.

## Possible Solution

## Steps to Reproduce the Problem

  1. using the VPN glorytun UDP
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 0.1023 5.4.81-mptcp
  - OpenMPTCProuter platform: x86_x64

Log evey hour : 
https://pastebin.com/t8eBLv5Z

all Log : 
https://pastebin.com/KmLeVJ0x

Ysurac, 2020-12-24:
I can reproduce the issue.
What is the result of ```glorytun-udp set``` via SSH or Services->Terminal ?

cayenne17, 2020-12-24:
> I can"t reproduce the issue.
> What is the result of `glorytun-udp set` via SSH or Services->Terminal ?

[image]

Ysurac, 2020-12-24:
I have same issue now...
The problem is maybe on VPS part, I update VPS script to set same kxtimeout setting.

cayenne17, 2020-12-30:
The issue is fix.

I have just tried to replace "/etc/glorytun-udp/post.sh" by
: https://github.com/Ysurac/openmptcprouter-vps/commit/6a0c3d32472833174f171018680e58def7b637a9

The issue as solved. 

I no longer lose connection to glorytun UDP VPN every hour

closed 2020-12-30 by cayenne17