Ysurac/openmptcprouter #1477: Prevent network config override

smallstech, 2020-12-25:
## Expected Behavior
Router should boot and allow connections from devices.

## Current Behavior
Router boots but no devices can connect (ethernet or wifi) when using openmptcprouter (built from master per guide in wiki)

## Specifications

  - OpenMPTCProuter version: Built from master (This issue didnt happen when built a few months ago I think around v.55)
  - OpenMPTCProuter VPS version: N/A
  - OpenMPTCProuter platform: Netgear R7800 (Tho this should effect the question)

Is there anyway when building from source to stop the openmptcppackage overwriting the network configuration to allow a initial boot placing a network config in source/files/etc/config seems to have no effect.

Ysurac, 2020-12-26:
I can't really help on this as it's not a supported device.
I think the only way would be to add a serial console.

smallstech, 2020-12-29:
Ah ok, so is there no way to stop openmptcprouter package overwriting files on build (or install)>

Ysurac, 2020-12-29:
No, there is no way to stop openmptcprouter to do what it have to do at initial boot and it's putting initial config. You have to modify that config if you know what is needed to make it work.

closed 2021-04-03 by github-actions[bot]