Ysurac/openmptcprouter #1478: Failed to update from 0.56.5 to 0.57.1 and backup broken from 0.56.5

xillibit, 2020-12-25:
I was running the router in version 0.56.5 and the VPS in 0.1022, i generated the backup on 0.56.5. Then i tried to update to 0.57.1, but when the router has rebooted it has broken. At this time it wasn't not possible to acces the web interface but the console on proxmox was still acessible.

When i done a new VM on proxmox by installing the router in version 0.57.1 it has workied untill i load the backup generated before. Then it was broken like above.

Now i have created a new VM with the version 0.56.5 with ext4 image and i loaded onto it a previous backup, with that it has worked. 

P.S. : i have updated now the VPS in 0.1023

Why the backup seems broken ? Why the update has failed from 0.56.5 to 0.57.1 ?

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: x86_64 VM on proxmox 6.3-3

Randdrick, 2020-12-25:
Hello,
it's not a bug, but a problem with your memory card and Squashfs image.
To resolve it and continue to use Squashfs image, make this operations

- Open a root connexion on your router with Putty
- Tape this command : `firstboot -y && reboot now`
- Restore your backup.

xillibit, 2020-12-25:
I have done a clean install with version 0.56.5 into a VM with the ext4 image but with that i haven't tried the update to 0.57.1

Ysurac, 2020-12-26:
Duplicate of https://github.com/Ysurac/openmptcprouter/issues/1468
Check this issue for the answer.

closed 2020-12-26 by Ysurac