Ysurac/openmptcprouter #1479: Sprint/Tmobile LTE Usb - Gateway down - no connection

codexual, 2020-12-25:
## Expected Behavior
Plug n play? 

## Current Behavior
So I have a USB LTE Modem Dongle (Netgear AirCard 341U) device. When I plug it into my Windows 10, and pull the IP configuration this is the result

Ethernet adapter Ethernet 5:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : NETGEAR WWAN Network Adapter
   Physical Address. . . . . . . . . : 00-A0-C6-00-00-18
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::b417:2fd:c50a:fe6a%46(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.168.2.4(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Lease Obtained. . . . . . . . . . : Friday, December 25, 2020 11:12:08 AM
   Lease Expires . . . . . . . . . . : Friday, December 25, 2020 1:12:07 PM
   Default Gateway . . . . . . . . . : 192.168.2.5
   DHCP Server . . . . . . . . . . . : 192.168.2.5
   DHCPv6 IAID . . . . . . . . . . . : 771793094
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-26-56-1F-B5-08-97-98-9F-24-3D
   DNS Servers . . . . . . . . . . . : 192.168.2.1
   NetBIOS over Tcpip. . . . . . . . : Enabled

So on the OpenMPTCProuter I go into System -> OpenMPTCProuter -> Then Settings Wizard Tab. I am able to get my other modems/devices to work (wan1, wan2, wan4) but not wan5, on wan5 I tried to do MacVLAN, as well with Normal using the Physical Interface as "wwan0" which is assigned by my USB Modem, I also setup a static IP (copied the same settings when I plugged into my windows 10 pc)

IPv4 Address: 192.168.2.4
IPv4 netmask: 255.255.255.0
IPv4 gateway 192.168.2.5

(I found it odd that the gateway is 192.168.2.5 by default, so I also tried 192.168.1.1 for the heck of it). I also put it as a DHCP Client, enable and disable "Multipath TCP", I read other users issues that's been posted on here. And try to figure it out myself before reaching out, but I am stuck and may I pretty please get some help? I will be posting additional logs below

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.57.1 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: null
  - OpenMPTCProuter platform: RPI4

codexual, 2020-12-25:
##System log

Fri Dec 25 19:43:39 2020 daemon.notice procd: /etc/rc.d/S90mptcp: Device 'bonding_masters' can't found!
Fri Dec 25 19:43:39 2020 daemon.notice procd: /etc/rc.d/S90mptcp: Use the hardware name like in ifconfig
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty55, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty55
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty55, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice unbound: default memory configuration
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: add serial interface ttyUSB1: event processed
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyUSB1, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty17, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty17
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty17, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty0
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty0, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty45, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty45
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty45, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty35, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty35
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty35, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty63, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty63
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty63, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 daemon.notice unbound: [9103:0] notice: init module 0: iterator
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty25, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty25
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty25, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Fri Dec 25 19:43:39 2020 daemon.info unbound: [9103:0] info: start of service (unbound 1.13.0).
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 daemon.info unbound: [9103:0] info: service stopped (unbound 1.13.0).
Fri Dec 25 19:43:39 2020 daemon.info unbound: [9103:0] info: server stats for thread 0: 1 queries, 0 answers from cache, 1 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Fri Dec 25 19:43:39 2020 daemon.info unbound: [9103:0] info: server stats for thread 0: requestlist max 0 avg 0 exceeded 0 jostled 0
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty9, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: add network interface wwan0: event processed
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=wwan0, subsystem=net
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty53, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: add cdc interface cdc-wdm0: custom event processed
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=cdc-wdm0, subsystem=usbmisc
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty15
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty15, subsystem=tty
Fri Dec 25 19:43:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Fri Dec 25 19:48:53 2020 daemon.info dnsmasq[13367]: using 1 more local addresses
Fri Dec 25 19:48:53 2020 daemon.info dnsmasq[13367]: read /etc/hosts - 4 addresses
Fri Dec 25 19:48:53 2020 daemon.info dnsmasq[13367]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Dec 25 19:48:53 2020 daemon.info dnsmasq-dhcp[13367]: read /etc/ethers - 0 addresses
Fri Dec 25 19:49:03 2020 user.notice post-tracking-post-tracking: wan5 (wan5) switched off because gateway down and ping from 192.168.2.4 error (4.2.2.1,8.8.8.8,80.67.169.12)

codexual, 2020-12-25:
## Kernal Logs

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83 (runner@fv-az128-843) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12288 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000]   Normal zone: 16384 pages used for memmap
[    0.000000]   Normal zone: 1048576 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:C1:CD:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 7943208K/8245248K available (7038K kernel code, 430K rwdata, 1840K rodata, 512K init, 539K bss, 236504K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000155] Console: colour dummy device 80x25
[    0.000394] printk: console [tty1] enabled
[    0.000430] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000452] pid_max: default: 32768 minimum: 301
[    0.000713] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000855] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001504] Disabling memory control group subsystem
[    0.002195] ASID allocator initialised with 32768 entries
[    0.002282] rcu: Hierarchical SRCU implementation.
[    0.002915] smp: Bringing up secondary CPUs ...
[    0.003301] Detected PIPT I-cache on CPU1
[    0.003351] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003787] Detected PIPT I-cache on CPU2
[    0.003821] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004287] Detected PIPT I-cache on CPU3
[    0.004319] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004388] smp: Brought up 1 node, 4 CPUs
[    0.004417] SMP: Total of 4 processors activated.
[    0.004430] CPU features: detected: 32-bit EL0 Support
[    0.004443] CPU features: detected: CRC32 instructions
[    0.004493] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004510] CPU: All CPU(s) started at EL2
[    0.004540] alternatives: patching kernel code
[    0.006073] devtmpfs: initialized
[    0.012977] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.013017] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014743] pinctrl core: initialized pinctrl subsystem
[    0.015029] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015623] NET: Registered protocol family 16
[    0.016985] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.017527] cpuidle: using governor ladder
[    0.017557] cpuidle: using governor menu
[... 755 lines cut ...]
[   30.664902] usb 1-1.3: USB disconnect, device number 8
[   31.012038] usb 1-1.3: new high-speed USB device number 9 using xhci_hcd
[   31.124900] usb 1-1.3: New USB device found, idVendor=04e8, idProduct=6863, bcdDevice= c.00
[   31.133278] usb 1-1.3: New USB device strings: Mfr=7, Product=8, SerialNumber=9
[   31.140599] usb 1-1.3: Product: SAMSUNG_Android
[   31.145139] usb 1-1.3: Manufacturer: SAMSUNG
[   31.149418] usb 1-1.3: SerialNumber: RF8M74J72CE
[... cut ...]
[   96.839539] usb 1-1.2: no of_node; not parsing pinctrl DT
[   96.842324] rndis_host 1-1.2:1.0: no of_node; not parsing pinctrl DT
[   96.854568] rndis_host 1-1.2:1.0 usb0: register 'rndis_host' at usb-0000:01:00.0-1.2, RNDIS device, 8a:89:c2:a5:7d:db
[  204.832339] bcmgenet fd580000.ethernet eth0: Link is Down
[  204.854297] bcmgenet: Skipping UMAC reset
[  204.859431] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[  204.866914] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

Ysurac, 2020-12-26:
This seems to be a QMI device. So you need to use QMI protocol in Network->Interface for this dongle. MacVLAN must not be used with this. You can refer to OpenWrt documentation/forum for more info (I don't have any QMI devices).

codexual, 2020-12-26:
Additionally is there a way to leave the settings alone, because every time I remove my other devices and plug them back in it changes the interface back to eth0, instead of leaving them as is.

closed 2021-04-01 by github-actions[bot]