Ysurac/openmptcprouter #1480: The router is restarting itself since 0.56+

WazoAkaRapace, 2020-12-25:
## Expected Behavior
The router have to work 24/7.

## Current Behavior
Since the 0.56 version, the router is restarting itself. It's not happening in 0.55. Tested on Rpi 2B and Rpi 3B.
Multiple SD card tested.

## Possible Solution

## Steps to Reproduce the Problem

  1. Install the router (0.56+) on a Rpi 2B or 3B
  2. Wait.

## Context (Environment)
1 Rpi as the router, 2 4G + 1 ADSL (issue happened with only one 4G + ADSL too)

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: RPI3 x86

## Logs

It look like soimething important fail at the end and crash the router...

> [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.83 (runner@fv-az32-410) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 46080 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30860 r8192 d22388 u61440
[    0.000000] pcpu-alloc: s30860 r8192 d22388 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240384
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x2aff8000-0x2eff8000] (64MB)
[    0.000000] Memory: 813828K/970752K available (6416K kernel code, 232K rwdata, 1688K rodata, 2048K init, 400K bss, 91388K reserved, 65536K cma-reserved, 118784K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2fc/0x430 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000020] Switching to timer-based delay loop, resolution 52ns
[    0.000187] Console: colour dummy device 80x30
[    0.000617] printk: console [tty1] enabled
[    0.000665] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000700] pid_max: default: 32768 minimum: 301
[    0.000894] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000929] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001801] Disabling memory control group subsystem
[    0.001842] CPU: Testing write buffer coherency: ok
[    0.002205] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002679] Setting up static identity map for 0x200000 - 0x200060
[... 754 lines cut ...]
[   10.103066] lowmem_reserve[]: 0 0 0 0
[   10.108005] DMA: 2*4kB (ME) 1*8kB (U) 3*16kB (UM) 3*32kB (ME) 5*64kB (UME) 4*128kB (M) 6*256kB (UM) 3*512kB (ME) 5*1024kB (M) 2*2048kB (M) 161*4096kB (M) = 672736kB
[   10.126670] HighMem: 37*4kB (UMC) 24*8kB (M) 10*16kB (UMC) 4*32kB (UMC) 2*64kB (MC) 3*128kB (UC) 2*256kB (UC) 1*512kB (U) 2*1024kB (UM) 1*2048kB (C) 35*4096kB (MC) = 149620kB
[   10.146495] 5193 total pagecache pages
[   10.151781] 242688 pages RAM
[   10.156132] 46080 pages HighMem/MovableOnly
[   10.161730] 22335 pages reserved
[   10.166224] 16384 pages cma reserved
[   10.175289] vmap allocation for size 65536 failed: use vmalloc=<size> to increase size
[   10.186434] vmap allocation for size 106496 failed: use vmalloc=<size> to increase size
[   10.197759] vmap allocation for size 106496 failed: use vmalloc=<size> to increase size
[   10.208819] vmap allocation for size 65536 failed: use vmalloc=<size> to increase size
[   10.220142] vmap allocation for size 73728 failed: use vmalloc=<size> to increase size
[   10.230972] kmodloader: 3 modules could not be probed
[   10.237461] kmodloader: - ath9k - 0
[   10.242172] kmodloader: - ath9k_htc - 0
[   10.247201] kmodloader: - rtl8192cu - 0
[   13.983963] 8021q: adding VLAN 0 to HW filter on device eth0
[   13.985228] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   13.994482] 8021q: adding VLAN 0 to HW filter on device wan1
[   14.011143] IPv6: ADDRCONF(NETDEV_CHANGE): wan1: link becomes ready
[   14.021301] 8021q: adding VLAN 0 to HW filter on device wan2
[   14.036474] 8021q: adding VLAN 0 to HW filter on device wan3
[   14.569993] EXT4-fs error (device mmcblk0p2): __ext4_new_inode:1114: comm uci: failed to insert inode 142: doubly allocated?
[   14.591008] EXT4-fs (mmcblk0p2): Remounting filesystem read-only

Ysurac, 2020-12-26:
You use the 32bits image ? Did you try the 64bits image ?

WazoAkaRapace, 2020-12-26:
32bits only, i'm going to try the 64bit right now. (I was waiting for a fresh new SD card to maximise the chances...)

WazoAkaRapace, 2020-12-26:
It up for an hour now, the log is not poping. It might be the 64 version or the fresh new SD Card. I think it was the SD card, maybe 2 broken in a row...

closed 2020-12-30 by Ysurac