Ysurac/openmptcprouter #1482: Portforward Multi IP's

MTSmhas, 2020-12-29:
Hi i have a problem. The Multi IP OMR settings works. But how can i forward Ports to the VPS?

I have read all postings but i have nothing found to resolve my problem.

MTSmhas, 2020-12-29:
I can go out with the second ip. But i can noting forward to the second ip.

Ysurac, 2020-12-29:
How did you configure port forward ?

MTSmhas, 2020-12-29:
[image]
[image]

MTSmhas, 2020-12-29:
[image]
But i think the /etc/shorewall/rules are wrong...

MTSmhas, 2020-12-29:
[image]
sry thats the correct screen.

MTSmhas, 2020-12-29:
But now all Ports closed.

MTSmhas, 2020-12-29:
But if I do the extern adresse to any is on all ip address the port 22 open.

MTSmhas, 2020-12-29:
If i set the external IP from the server i have the same error

MTSmhas, 2020-12-29:
[image]

MTSmhas, 2020-12-29:
[image]
[image]
[image]
This is my omrip1_rule

Ysurac, 2020-12-29:
The correct way is to set external VPS ip as external IP., like this seems to be in last shorewall screen.
Shadowsocks is not used in redirection from VPS.

Can you ping from the VPS 10.255.249.6 ? If yes, what do you have using tcpdump on router side on the interface with this IP ?

MTSmhas, 2020-12-29:
the ping from the vps is not working
[image]

MTSmhas, 2020-12-29:
but i can ping the 192.168.1.13... 
[image]

Ysurac, 2020-12-29:
ok, so it's the problem, the gre tunnel is not working.
What is the result of ```ip a``` on the router ?

Ysurac, 2020-12-29:
And what is the VPN used ? (only glorytun TCP is tested for now)

MTSmhas, 2020-12-29:
I use glorytun TCP

MTSmhas, 2020-12-29:
root@USA-Router:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group                                                                                                                                                              default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen                                                                                                                                                              1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 10                                                                                                                                                             00
    link/sit 0.0.0.0 brd 0.0.0.0
4: eth0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel                                                                                                                                                              state UP group default qlen 1000
    link/ether 00:0c:29:03:d4:89 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.154/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fddb:1d41:bbb3::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::20c:29ff:fe03:d489/64 scope link
       valid_lft forever preferred_lft forever
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 10                                                                                                                                                             00
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN grou                                                                                                                                                             p default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN grou                                                                                                                                                             p default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen                                                                                                                                                              1000
    link/gre6 :: brd ::
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
10: wan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP                                                                                                                                                              group default qlen 100
    link/ether 5a:9a:c8:e3:f2:e9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.114/24 brd 192.168.1.255 scope global wan1
       valid_lft forever preferred_lft forever
11: gre4-omrip3gre@NONE: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 128                                                                                                                                                             0 qdisc noqueue state UNKNOWN group default qlen 50
    link/gre 10.255.255.2 peer 206.221.178.45
    inet 10.255.249.14/30 brd 10.255.249.15 scope global gre4-omrip3gre
       valid_lft forever preferred_lft forever
17: gre4-omrip0gre@NONE: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 128                                                                                                                                                             0 qdisc noqueue state UNKNOWN group default qlen 50
    link/gre 10.255.255.2 peer 206.221.178.44
    inet 10.255.249.2/30 brd 10.255.249.3 scope global gre4-omrip0gre
       valid_lft forever preferred_lft forever
19: gre4-omrip2gre@NONE: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 128                                                                                                                                                             0 qdisc noqueue state UNKNOWN group default qlen 50
    link/gre 10.255.255.2 peer 206.221.178.43
    inet 10.255.249.10/30 brd 10.255.249.11 scope global gre4-omrip2gre
       valid_lft forever preferred_lft forever
20: gre4-omrip4gre@NONE: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 128                                                                                                                                                             0 qdisc noqueue state UNKNOWN group default qlen 50
    link/gre 10.255.255.2 peer 206.221.178.46
    inet 10.255.249.18/30 brd 10.255.249.19 scope global gre4-omrip4gre
       valid_lft forever preferred_lft forever
21: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_code                                                                                                                                                             l state UNKNOWN group default qlen 1000
    link/none
[... cut ...] 0
       valid_lft forever preferred_lft forever
22: gre4-omrip1gre@NONE: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 128                                                                                                                                                             0 qdisc noqueue state UNKNOWN group default qlen 50
    link/gre 10.255.255.2 peer 206.221.178.42
    inet 10.255.249.6/30 brd 10.255.249.7 scope global gre4-omrip1gre
       valid_lft forever preferred_lft forever

MTSmhas, 2020-12-29:
[image]

Ysurac, 2020-12-29:
Try to ping from the router 10.255.249.5, this should put gre tunnel up (this will be solved in next release).
Then check from VPS if you can ping 10.255.249.6 again, and check if port forward work or not.

MTSmhas, 2020-12-29:
router:~# ping 10.255.249.5
PING 10.255.249.5 (10.255.249.5) 56(84) bytes of data.
64 bytes from 10.255.249.5: icmp_seq=1 ttl=64 time=0.123 ms
64 bytes from 10.255.249.5: icmp_seq=2 ttl=64 time=0.096 ms

MTSmhas, 2020-12-29:
from router i can ping 10.255.249.5 this is already okay but i cant ping from vps the 10.255.249.6

MTSmhas, 2020-12-29:
nmap -p22 -Pn 10.255.249.6
Starting Nmap 7.70 ( https://nmap.org ) at 2020-12-29 11:56 EST
Nmap scan report for 10.255.249.6
Host is up.

PORT   STATE    SERVICE
22/tcp filtered ssh

Nmap done: 1 IP address (1 host up) scanned in 2.30 seconds

MTSmhas, 2020-12-29:
i think the firewall is the problem.

MTSmhas, 2020-12-29:
But the curl ifconfig.me 206.221.178.42 is correct.

MTSmhas, 2020-12-29:
oh it works... i have stoip the ss.rules and start again now works.

MTSmhas, 2020-12-29:
Okay i hope you can fix this in the next release. But Thanks for all!!!!

MTSmhas, 2020-12-29:
stop.... only the port in vpn is open not in the vps..

MTSmhas, 2020-12-29:
so now ping run and ports closed... i dont understand thsi.

MTSmhas, 2020-12-29:
okay i found the error. when i disable the ss.rules work the port forwarding but the ip from client change to the standart vps ip. I think the rules are not realy okay..

Ysurac, 2020-12-29:
I updated the VPS script, instead of using GRE tunnel, this use the VPN for now.
Not ideal, but working for now.

MTSmhas, 2020-12-30:
Okay can you tell me how the ip config in the vps is. I set the ips with eth0:1 eth0:2 etc

MTSmhas, 2020-12-30:
Now works without errors. I have the vps reinstall and the ips to the eth0 added "ip addr add 10.1.102.60No works without inuse. I have the vps reinstall and the ips to the eth0 added "ip addr add 10.1.102.60/24 dev eth0" without a second eth0:0.

closed 2020-12-30 by MTSmhas