Ysurac/openmptcprouter #1483: Ajouter les ip de mes appareils 4G automatiquement

felixferr, 2020-12-29:
Bonjour, je souhaiterais envoyer mes adresses ip de mes appareils 4G de façon automatique sur wan1, wan2 etc ... en fonction du nombre d'appareils que je connecte. Auriez vous des pistes de comment je peux réaliser cela, si c'est possible ?

Ysurac, 2020-12-29:
Je ne comprend pas trop ce que tu veux faire.
Les interfaces peuvent être configurées pour recevoir les adresses via DHCP, mais uniquement si le routeur possède le nombre de ports physique suffisant (ou via USB pour les clefs).

felixferr, 2020-12-29:
Effectivement c'est pas très clair, je vais tenter de l'être cette fois ci.
Je souhaite que au lieu de paramétrer manuellement les adresses ip de mes téléphones 4G ici : [image]

,que ceci se fasse de manière automatisé. Du genre à partir l'ip mon téléphone 4G se mette sur wan1 sans le faire à la main.

Ysurac, 2020-12-29:
Si wan1 est configuré en client DHCP et que le téléphone a un serveur DHCP (ce qui est au final souvent le cas) alors les IPs sont configurées automatiquement.

felixferr, 2020-12-29:
En fait quand je dis paramétrer, c'est sans copier/coller les adresses ip dans les cases wan1, wan2 mais que je puisse le faire avec un script.

closed 2020-12-29 by felixferr