Ysurac/openmptcprouter #1484: bridage a 47mbps

Aur7, 2020-12-29:
Bonjour;
j'ai installer openmptcprouter depuis deux jours sur un serveur Windows 2019 dans une VM sous VMWare workstation 16 pro avec l'image en OVA. mon serveur distant est un VPS STD-2 de chez Pulseheberg.

Je rencontre donc un problème, j'ai connecter un routeur 4/5G qui auparavant moyennait un débit descendant entre 80 et 150 mbps. la je me retrouve avec un débit toujours entre 45 et 47 mbps, en speedtest on ressent vraiment un bridage. ma question: mais de ou vient ce bridage ?!
mon VPS en speedtest donne 290mbps, ram et CPU sons loin d'aitre a saturation, ma VM a 2GO de ram et 8Vcore, la carte réseau est bien configuré en illimitée
j'ai chercher sans trouver ni indice ni solution :/

[Dernière précision même si je pense que ce n'est pas lier, dans l’état du système j'ai sa au niveau du serveur https://i.imgur.com/gHnWlMP.png ]

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: 
Merci pour vautre aide, Cordialement Aur7

Aur7, 2020-12-29:
je re préside 
OpenMPTCProuter version: v0.57.1 r0+15225-bfc433efd4
OpenMPTCProuter VPS version: 0.1023.5.4.81
OpenMPTCProuter platform: 64 bit
(qui on disparut)

Ysurac, 2020-12-29:
Le fait qu'il n'arrive pas à contacter l'API du serveur n'est pas une bonne chose. Il faut voir dans le journal système la raison et éventuellement dans /var/log/daemon.log sur le VPS si il n'y a pas d'erreurs.
Il me faudrait aussi le load average côté VPS et routeur lors d'un speedtest.

Aur7, 2020-12-29:
Merci pour cette réponse rapide, 
voici le fichier daemon.log 
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/5750937/daemon.log)
je voit pour faire le load average.
Merci

Aur7, 2020-12-29:
le load AVG 
serveur VPS : `15:07:04 up 2 days,  1:11,  1 user,  load average: 0.00, 0.00, 0.00
root@Pasrel-VPS:~# uptime
 15:07:06 up 2 days,  1:11,  1 user,  load average: 0.08, 0.02, 0.01
root@Pasrel-VPS:~# uptime
 15:07:09 up 2 days,  1:11,  1 user,  load average: 0.08, 0.02, 0.01
root@Pasrel-VPS:~# uptime
 15:07:12 up 2 days,  1:11,  1 user,  load average: 0.07, 0.02, 0.00
root@Pasrel-VPS:~# uptime
 15:07:15 up 2 days,  1:11,  1 user,  load average: 0.07, 0.02, 0.00
root@Pasrel-VPS:~# uptime
 15:07:17 up 2 days,  1:11,  1 user,  load average: 0.07, 0.02, 0.00
root@Pasrel-VPS:~# uptime
 15:07:19 up 2 days,  1:11,  1 user,  load average: 0.07, 0.02, 0.00
root@Pasrel-VPS:~# uptime
 15:07:22 up 2 days,  1:11,  1 user,  load average: 0.06, 0.02, 0.00`
VM routeur local : https://i.imgur.com/cjIpslJ.png

Ysurac, 2020-12-29:
Comment est fait le speedtest ?
Le résultat de ```omr-test-speed``` via SSH ou Services->Terminal sur le routeur ?

Aur7, 2020-12-29:
sur le VPS et un serv local ubuntu en local avec CLI de speedtest by Ookla, en SSH
et sur mes PC et serveur Windows avec l'app speedtest by Ookla.
(je ne connait pas `omr-test-speed`)

Aur7, 2020-12-29:
Je vien de comprend, j'ai taper `omr-test-speed` en SSH, la j’attends le résultat mais sa paret extrêmement long

Aur7, 2020-12-29:
voici le début du test

[image]

Ysurac, 2020-12-29:
Ctrl+C pour arrêter.

Aur7, 2020-12-29:
oui pas de soucies, donc sa aide a savoir ce qu'il se passe ? ^^

Ysurac, 2020-12-29:
Ce qui donne donc une vitesse de 4.3Mo/s, donc 34.4Mb/s, ce qui est cohérent avec ton précédent speedtest.
Il est possible de faire un ```omr-test-speed wan1``` pour tester la vitesse sur l'interface réelle wan1. ```ip a``` pour voir les interfaces réelles.

Aur7, 2020-12-29:
pour `omr-test-speed wan1` sa me donne sa :
[image]

et pour ip a :

[image]
[image]

Aur7, 2020-12-29:
Y'a t'il moyen de faire un teste local ? (entre un de mes PC et le routeur) ?
voir si c'est un souci lié a la VM

Ysurac, 2020-12-29:
Justement c'est ce qu'à fait la dernière commande. On a donc 5.7Mo/s, donc 45.6Mb/s entre OMR et ton routeur 4G.

Aur7, 2020-12-29:
ha ba c'est la le problème !

Ysurac, 2020-12-29:
oui. Après la vitesse en 4G peut être très variable.

Aur7, 2020-12-29:
j'ai de la TRES bonne 4G 50mbps c'est le minimum, je tourne entre 80 et 150 habituellement même en jours pluvieux 
[image]

Aur7, 2020-12-29:
Avant d'installer OMR
[image]
après 
[image]
faut avouer que c'est troublant x)

Ysurac, 2020-12-29:
Week end puis semaine. Avec le télétravail il y a plus d'utilisateurs sur les antennes.

Aur7, 2020-12-29:
a ba si tu veut je désactive tout, je branche mon HUAWEI 5G CPE Win a mon PC et on voit x)

Aur7, 2020-12-29:
Alors, je ne sais pas si c'est le hasard ou quoi, mais d'avoir tout débrancher, dé paramétré, tester, re paramétré et re tester.
et ba sa a solutionner mon problème.
je débranche tout, test 80mbps, je rebranche tout et teste a nouveau 80.
Deloser de t'avoir mobiliser alors que la réponse était "a ba fallait tout re démarré" xD

Par-contre, je fait quoi pour le souci d'API avec le serveur ?
et un grand merci au passage pour ton travaille !

Ysurac, 2020-12-30:
Pour l'API, il faut voir sur le routeur, dans Etat->Journal Système ce qu'il arrive à faire avec le VPS.

Aur7, 2021-01-01:
Bonjour, voici les log du 1er janvier (bonne année)
`Fri Jan  1 00:00:00 2021 cron.err crond[5685]: USER root pid 13182 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 00:08:01 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21
Fri Jan  1 00:08:01 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21 AUR7-SERVER
Fri Jan  1 00:08:01 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 00:33:42 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.210 98:d8:63:c0:1d:8c
Fri Jan  1 00:33:42 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.210 98:d8:63:c0:1d:8c HF-LPT230
Fri Jan  1 00:33:42 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 00:44:34 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.101 94:e4:ba:9c:33:a0
Fri Jan  1 00:44:34 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.101 94:e4:ba:9c:33:a0 04f9447d29f095e9554eaa3fa05f0a7
Fri Jan  1 00:44:34 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 01:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.211 1c:3b:f3:c6:db:10
Fri Jan  1 01:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.211 1c:3b:f3:c6:db:10 TL-WR820N
Fri Jan  1 01:16:03 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 01:38:12 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 01:38:53 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 02:00:00 2021 cron.err crond[5685]: USER root pid 30036 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 02:00:00 2021 cron.err crond[5685]: USER root pid 30037 cmd /etc/init.d/omr-bypass bypass_asn
Fri Jan  1 02:37:10 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.158 d0:7f:a0:b0:72:b1
Fri Jan  1 02:37:10 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.158 d0:7f:a0:b0:72:b1 Galaxy-Tab-A-2016
Fri Jan  1 02:37:10 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 03:39:28 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 03:40:08 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 04:00:00 2021 cron.err crond[5685]: USER root pid 14145 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 05:40:45 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 05:41:25 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 05:42:37 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21
Fri Jan  1 05:42:37 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21 AUR7-SERVER
Fri Jan  1 05:42:37 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:00:00 2021 cron.err crond[5685]: USER root pid 30044 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 06:00:25 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.101 94:e4:ba:9c:33:a0
Fri Jan  1 06:00:25 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.101 94:e4:ba:9c:33:a0 04f9447d29f095e9554eaa3fa05f0a7
Fri Jan  1 06:00:25 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:30:00 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.210 98:d8:63:c0:1d:8c
Fri Jan  1 06:30:00 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.210 98:d8:63:c0:1d:8c HF-LPT230
Fri Jan  1 06:30:00 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:41:37 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 06:42:17 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 07:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.211 1c:3b:f3:c6:db:10
Fri Jan  1 07:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.211 1c:3b:f3:c6:db:10 TL-WR820N
Fri Jan  1 07:16:03 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 08:00:00 2021 cron.err crond[5685]: USER root pid 13795 cmd /etc/init.d/vnstat_backup backup
[... cut ...]
Fri Jan  1 11:51:11 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.180 00:d8:61:03:83:ad
Fri Jan  1 11:51:11 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.180 00:d8:61:03:83:ad LAPTOP-Aur7
Fri Jan  1 11:51:11 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 11:52:49 2021 daemon.err uhttpd[6393]: luci: accepted login on / for root from 192.168.100.180`

Ysurac, 2021-01-01:
Comme précisé dans le template, je ne télécharge pas de fichier car ça rend le suivis de bug impossible.

Aur7, 2021-01-01:
ba pas de soucies : 
`Fri Jan  1 00:00:00 2021 cron.err crond[5685]: USER root pid 13182 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 00:08:01 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21
Fri Jan  1 00:08:01 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21 AUR7-SERVER
Fri Jan  1 00:08:01 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 00:33:42 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.210 98:d8:63:c0:1d:8c
Fri Jan  1 00:33:42 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.210 98:d8:63:c0:1d:8c HF-LPT230
Fri Jan  1 00:33:42 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 00:44:34 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.101 94:e4:ba:9c:33:a0
Fri Jan  1 00:44:34 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.101 94:e4:ba:9c:33:a0 04f9447d29f095e9554eaa3fa05f0a7
Fri Jan  1 00:44:34 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 01:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.211 1c:3b:f3:c6:db:10
Fri Jan  1 01:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.211 1c:3b:f3:c6:db:10 TL-WR820N
Fri Jan  1 01:16:03 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 01:38:12 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 01:38:53 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 02:00:00 2021 cron.err crond[5685]: USER root pid 30036 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 02:00:00 2021 cron.err crond[5685]: USER root pid 30037 cmd /etc/init.d/omr-bypass bypass_asn
Fri Jan  1 02:37:10 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.158 d0:7f:a0:b0:72:b1
Fri Jan  1 02:37:10 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.158 d0:7f:a0:b0:72:b1 Galaxy-Tab-A-2016
Fri Jan  1 02:37:10 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 03:39:28 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 03:40:08 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 04:00:00 2021 cron.err crond[5685]: USER root pid 14145 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 05:40:45 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 05:41:25 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 05:42:37 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21
Fri Jan  1 05:42:37 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.196 ac:1f:6b:e9:6c:21 AUR7-SERVER
Fri Jan  1 05:42:37 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:00:00 2021 cron.err crond[5685]: USER root pid 30044 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 06:00:25 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.101 94:e4:ba:9c:33:a0
Fri Jan  1 06:00:25 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.101 94:e4:ba:9c:33:a0 04f9447d29f095e9554eaa3fa05f0a7
Fri Jan  1 06:00:25 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:30:00 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.210 98:d8:63:c0:1d:8c
Fri Jan  1 06:30:00 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.210 98:d8:63:c0:1d:8c HF-LPT230
Fri Jan  1 06:30:00 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 06:41:37 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jan  1 06:42:17 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jan  1 07:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.211 1c:3b:f3:c6:db:10
Fri Jan  1 07:16:03 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.211 1c:3b:f3:c6:db:10 TL-WR820N
Fri Jan  1 07:16:03 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 08:00:00 2021 cron.err crond[5685]: USER root pid 13795 cmd /etc/init.d/vnstat_backup backup
Fri Jan  1 08:02:31 2021 daemon.err /usr/bin/ss-redir[22196]: server recv: Operation timed out
[... cut ...]
Fri Jan  1 11:51:11 2021 daemon.info dnsmasq-dhcp[15890]: DHCPREQUEST(eth0) 192.168.100.180 00:d8:61:03:83:ad
Fri Jan  1 11:51:11 2021 daemon.info dnsmasq-dhcp[15890]: DHCPACK(eth0) 192.168.100.180 00:d8:61:03:83:ad LAPTOP-Aur7
Fri Jan  1 11:51:11 2021 daemon.warn dnsmasq-dhcp[15890]: not giving name AUR7-SERVER.lan to the DHCP lease of 192.168.100.196 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.195
Fri Jan  1 11:52:49 2021 daemon.err uhttpd[6393]: luci: accepted login on / for root from 192.168.100.180`

Ysurac, 2021-01-01:
I don't see error, should be related to openmptcprouter-vps. Maybe it's only taking too much time for status page.

Aur7, 2021-04-02:
how do I do it? : /

closed 2021-07-08 by github-actions[bot]