Ysurac/openmptcprouter #1486: OMR-Bypass still appears to use master interface

alexciarlillo, 2020-12-30:
## Expected Behavior
When using omr-bypass traffic should only go over the specified interface.

## Current Behavior
When using omr-bypass it appears to ignore the selected interface and still use the master interface.

## Possible Solution
Unknown

## Steps to Reproduce the Problem

  1. Have at least 2 gateway/interfaces setup.
  2. Specify a device use the non-master gateway either via MAC Address or IP Address (this happens on both)
  3. Give a few minutes to ensure the traffic should be flowing accordingly.
  4. Use an external IP detection service to see the reported IP does not match the selected gateway's WAN IP.
  5. Additionally, perform a speed test and verify traffic is still flowing over the master interface.

## Context (Environment)

Here are some screenshots which show my configuration and validate what I am saying. One thing to note: the WAN address shown in the network overview do not match the external address shown by the device.

Another thing to note: when I enabled the bypass, I cannot even reach the intended gateway (192.168.4.1) but I can still reach the master (192.168.6.1). Once I disable the bypass I can once again access it.

Bypass configuration along w/ my device properties:
[image]

Network overview:
[image]

Traffic graph and recognized external IP while on bypass:
[image]

## Specifications

  - OpenMPTCProuter version: 0.57.1 (also on 0.56.5)
  - OpenMPTCProuter VPS version: 0.1023 (also on v0.1022)
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-01-04:
Maybe a problem when using VLAN interface.
Can you give me the result of ```iptables-save```, ```ip a``` and ```ip r``` from the router via SSH or Services->Terminal when bypass is enabled ?

alexciarlillo, 2021-01-05:
Sure thing!

`ip a`
```
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
3: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 40:62:31:11:e9:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd5b:85ad:2350::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::4262:31ff:fe11:e99e/64 scope link 
       valid_lft forever preferred_lft forever
5: eth1: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 40:62:31:11:e9:9f brd ff:ff:ff:ff:ff:ff
6: eth2: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 40:62:31:11:e9:a0 brd ff:ff:ff:ff:ff:ff
7: eth3: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 40:62:31:11:e9:a1 brd ff:ff:ff:ff:ff:ff
8: gre0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
9: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
10: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
12: bond0: <BROADCAST,MULTICAST,MASTER,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 76:82:ed:88:c4:fd brd ff:ff:ff:ff:ff:ff
13: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
15: eth0.4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 40:62:31:11:e9:a1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.40/24 brd 192.168.4.255 scope global eth0.4
       valid_lft forever preferred_lft forever
16: eth0.6@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 100
    link/ether 40:62:31:11:e9:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.6.60/24 brd 192.168.6.255 scope global eth0.6
       valid_lft forever preferred_lft forever
18: eth0.10@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 40:62:31:11:e9:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.10/24 brd 192.168.10.255 scope global eth0.10
       valid_lft forever preferred_lft forever
22: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 100
    link/none 
    inet 10.255.254.2 peer 10.255.254.1/32 scope global tun0
       valid_lft forever preferred_lft forever
25: ifb4tun0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether ae:2c:52:7a:06:ec brd ff:ff:ff:ff:ff:ff
```

`ip r`
```
default via 192.168.6.1 dev eth0.6 
default via 192.168.4.1 dev eth0.4 metric 6 
default via 192.168.6.1 dev eth0.6 metric 7 
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2 
10.255.254.2 dev tun0 scope link metric 1200 
45.79.157.40 via 192.168.6.1 dev eth0.6 metric 1 
45.79.157.40 via 192.168.4.1 dev eth0.4 metric 6 
45.79.157.40 via 192.168.6.1 dev eth0.6 metric 7 
[... 616 lines cut ...]
-A zone_wan_dest_ACCEPT -o eth0.4 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth0.6 -m conntrack --ctstate INVALID -m comment --comment "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth0.6 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o eth0.4 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth0.6 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
[... cut ...]
-A zone_wan_src_REJECT -i eth0.4 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth0.6 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Tue Jan  5 13:37:42 2021
```

Besides a few port-forwards, the only other non-standard thing (besides using VLANs for the gateways) was adding an additional VLAN (10) for some IoT devices on my Wifi which is managed by my Uibiquiti Unifi Controller. However, my PC where I am testing from is hard-wired in and is not on that VLAN so I do not believe that is relevant.

Ysurac, 2021-01-06:
I don't see error here.
Can I have the result of ```uci show omr-bypass``` on the router via SSH or Services->Terminal ?

alexciarlillo, 2021-01-06:
```
root@KENNEDY:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.eth0=interface
omr-bypass.eth0.id='2'
omr-bypass.eth0.10='interface'
omr-bypass.eth0.4='interface'
omr-bypass.eth0.6='interface'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.mlvpn0=interface
omr-bypass.mlvpn0.id='3'
omr-bypass.@lan_ip[0]=lan_ip
omr-bypass.@lan_ip[0].interface='eth0.6'
omr-bypass.@lan_ip[0].ip='192.168.100.113'
omr-bypass.@lan_ip[0].note='Alex Pixel'
omr-bypass.@macs[0]=macs
omr-bypass.@macs[0].mac='FA:83:2A:00:F1:C2'
omr-bypass.@macs[0].note='Alex iPad'
omr-bypass.@macs[0].interface='eth0.4'
omr-bypass.@macs[1]=macs
omr-bypass.@macs[1].mac='00:D8:61:15:AD:28'
omr-bypass.@macs[1].note='Alex PC'
omr-bypass.@macs[1].interface='eth0.4'
```

All looks right to me as well. It is very strange. I just re-confirmed again the behavior is the same and it is currently sending traffic over the eth0.6 interface.

Ysurac, 2021-01-06:
I found the issue, I will fix it.

alexciarlillo, 2021-01-06:
That's great! Thank you! For my own curiosity would you mind sharing what the issue was or how you were able to detect it from the information I shared? If it is too in depth don't worry about it though. Thanks again for the quick response, I really appreciate this project.

Ysurac, 2021-01-06:
When I add interface I didn't check the name, and omr-bypass.eth0.6 is a problem, I will replace that with omr-bypass.eth0_6 so I can write interface id here.

redkite127, 2021-01-17:
Happy you found the issue, I'm exactly in the same situation :+1:

redkite127, 2021-04-18:
UP
(we are still expecting the bug fix that @Ysurac  mentionned early January :grimacing:)
I'm happy to help/test/modify my installation if that can help :+1:

redkite127, 2021-04-25:
[ v0.58rc1](https://github.com/Ysurac/openmptcprouter/issues/1619) solved this issue for me, it's working again! thanks :+1:

closed 2021-07-29 by github-actions[bot]