Ysurac/openmptcprouter #1487: Testing with a MOFI4500 set up as WiFi Client

topbilling, 2020-12-30:
## Expected Behavior
Connect MOFI4500 as WAN1

## Current Behavior
Very slow speed with MOFI4500 connected via WiFi client

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 0.1023 5.4.81
  - OpenMPTCProuter platform: RPI4B

I'm trying to set up a test bench before I deploy to my full network.  The test bench consists of a Raspberry Pi 4B bonding a WG1608 with a VZW LTE plan and a MOFI4500 which is setup as a WiFI bridge connected to my normal home WiFi.  I'm waiting on a few additional SIM cards which I can put into the MOFI in order to test bonding two different LTE connections.  In the meantime, I'm trying to get the initial test setup and the MOFI (in WiFi bridge mode) either doesn't connect or connects with very slow speeds.  I suspect it's due to DHCP conflict, but I've tried putting the MOFI in pass through, setting up in DMZ, etc. and nothing works.  Is there something obvious I'm missing or is the interim test configuration not possible?

Thanks!

Ysurac, 2021-01-04:
I don't really understand your configuration. I would need a screenshot of OpenMPTCProuter status page.

topbilling, 2021-01-05:
I flashed the MOFI with Rooter and set it up to connect via WiFi tethering to iPhone on a prepaid AT&T plan.  The other connection is unchanged... a WG1608 with a VZW prepaid plan.  Everything is up and running normally except the VZW connection is very slow when downloading.  It's seems to be uploading at normal speeds.  I'm testing during non-congested times.  Also, I've tested the connection without OpenMPTCP and it works fine.  Any ideas why the download on one connection would be slower than expected?
[image]
[image]

closed 2021-04-11 by github-actions[bot]