Ysurac/openmptcprouter #1488: Aucune IP définie WAN

JeedHome44, 2020-12-31:
## Expected Behavior

 bonjour,
Je viens juste de faire un redémarrage de OMR et ça me dit qu'une IP n'est définie pour mes 2 WAN. Pourtant, je n'ai rien changé à la configuration à part avoir mis tous mes baux DHCP en infinite.

## Current Behavior

 c'est à cause des baux en non limité qu'il y a un problème?

J'ai bien vérifié les paramètres et les IP de chaque wan n'ont pas bougé et sont bien renseignés.

## Specifications

  - OpenMPTCProuter version:  0.56.5
  - OpenMPTCProuter VPS version:  0.1022
  - OpenMPTCProuter platform: 
RPi 4B 2 Go

JeedHome44, 2020-12-31:
Je viens d'appeler Free pour être sur et mon IP interne Freebox n'a pas changé, est toujours 192.168.0.1 mais impossible d'y accéder sur Chrome. Je pige pas.
Et impossible d'accéder à l'interface du routeur Huawei en 192.168.8.1
Y a un problème sur OMR. C'est en redémarrant mais lequel...

JeedHome44, 2020-12-31:
Je viens de mettre à jour mais toujours même problème

JeedHome44, 2020-12-31:
J'ai pleins de lignes dans les logs comme ça:

daemon.err unbound: [12556:0] error: udp connect failed: Network unreachable 192.X.X.X port 53

JeedHome44, 2020-12-31:
J'ai l'impression qu'une ligne dans l'interface LAN à été enlevé.

Par défaut il y a quelque chose dans Passerelle IPV4 et Diffusion IPV4?

JeedHome44, 2020-12-31:
J'ai l'impression que OMR n'arrive plus à voir les places autre que 100.X. J'ai du virer un truc dans l'interface LAN sans m'en rendre compte mais quoi...

Ysurac, 2020-12-31:
Pour ça il me faudrait la config de la partie network. Donc dans Système->OpenMPTCProuter le dernier onglet (juste la partie network) ou via SSH ou Services->Terminal ```uci show network```

JeedHome44, 2020-12-31:
La c'est même pire...
J'ai voulu remettre un BackUp et ça m'a mis une erreur main et interface OMR impossible à accéder.
J'ai donc remis la dernière mise à jour OMR sur la carte SD j'ai voulu remettre le BackUp et pareil.
Donc la, encore remis la mise à jour OMR et sans remettre de BackUp, j'ai juste configuré les 2 WAN et le VPS dans Assistant de configuration.
Les 2 wan et le VPS sont passés temporairement au vert et une minute après, encore au rouge. J'ai rien fait pourtant que mon LAN. J'ai rien déconnecté.
Je pige pas pourquoi ça bloque avec une mise à zéro propre...

JeedHome44, 2020-12-31:
La carte SD peut être être la case du problème?
Elle est morte?

JeedHome44, 2020-12-31:
C'est quoi le symptôme d'une carte SD grillé avec OMR? J'arrive à accéder à l'interface de OMR la sans BackUp donc je pige pas

JeedHome44, 2020-12-31:
Mais sans OMR je dois pouvoir accéder à Freebox OS sur le 192.168.0.1 non? Ou le 192.168.8.1 via Chrome?
C'est indépendant?

JeedHome44, 2020-12-31:
Log actuels qui sont les mêmes que j'avais avec ma configurations que j'ai fait en backup:

JeedHome44, 2020-12-31:
j'ai juste fait tout à l'heure un redémarrage de l'interface LAN puis un redémarrage de OMR et c'est là que tout est parti en vrille... mais je ne pige pas du tout pourquoi je ne peux pas aller sur les 2 interface de mes 2 wan... même OMR éteitn je devrait pouvoir accèder à Freebox OS et l'interface Huawei non?

JeedHome44, 2020-12-31:
[image]

JeedHome44, 2020-12-31:
Ce n'est pas la SD, j'ai essayé avec une autre et pareil... là c'est une capture d'écran avec système installer sur SD vierge et juste les WAN et VPS renseigné... Je pige pas... et le backup ne veut rien savoir....

JeedHome44, 2020-12-31:
[image]

JeedHome44, 2020-12-31:
ca fait ca sur le navigateur si j'essaie de restaurer n'importe quelle sauvegarde même une de novembre ou tout allait bien...

JeedHome44, 2020-12-31:
donc j'ai perdu tout... configuration ip fixe de plus de 100 appareil, tous les by-pass et tout... beau 31 décembre la... j'suis vert... juste à cause d'un reboot... je pige pas

Ysurac, 2020-12-31:
Alors déjà faudrait répondre quand je demande des renseignements ;)
Pour l'erreur avec luci : https://github.com/Ysurac/openmptcprouter/issues/1468

Kalimeiro, 2020-12-31:
Bonjour JeedHome44,

J'ai eu le même problème avec le message d'erreur sur LUCI en 0.57.1 et toujours d'actualité en 0.57.2 mais ce n'est pas le sujet ici, il faut être informaticien avec de l’expérience et de la logique pour trouver une solution de manière autonome.

Il est inutile d'appeler Free concernant le changement d'une IP, surtout quand elle est locale et que tu peux paramétrer toi même comme tu le souhaite cette IP depuis l'interface de ta freebox via ton navigateur, ton téléphone via l'application spécifique de Free, et même depuis l'interface d'administration disponible sur le site de free.fr

Le projet OMR est encore en phase BETA, de mémoire il n'y a qu'un seul développeur, pour ma part j'apporte mon soutiens comme je le peux afin de debug et soutenir un projet prometteur.

dans tes 17 messages en moins de 3 heures, afin de trouver une solution rapidement, je te suggère de : 

1°) Répondre à la demande de Ysurac : "Pour ça il me faudrait la config de la partie network. Donc dans Système->OpenMPTCProuter le dernier onglet (juste la partie network) ou via SSH ou Services->Terminal uci show network"

2°) retourner dans une ancienne version qui fonctionnait avec ton matériel, restaurer une sauvegarde fonctionnel, et si c'est OK je te suggère fortement de ne surtout plus toucher à rien.

Je suis désolé de cette intervention, mais ce n'est pas la première fois que j'ai remarqué tes multiples messages en mode SPAM et que je trouve personnellement désagréable à lire.

Avec plaisir de t'apporter des réponses techniques logiques, et bonne fête de fin d'année.

JeedHome44, 2020-12-31:
> Alors déjà faudrait répondre quand je demande des renseignements ;)
> 
> Pour l'erreur avec luci : https://github.com/Ysurac/openmptcprouter/issues/1468

Je n'ai pas répondu car je n'ai plus accès au système d'origine vu que j'ai remis à zéro. Et vu qu'en mettant le BackUp, j'ai l'erreur Luci, je ne peux donc rien faire.
Je ne comprends pas bien l'anglais et je ne sais pas comment résoudre le problème Luci. Je verrai demain. Merci pour la réponse aujourd'hui.

JeedHome44, 2020-12-31:
> Bonjour JeedHome44,
> 
> 
> 
> J'ai eu le même problème avec le message d'erreur sur LUCI en 0.57.1 et toujours d'actualité en 0.57.2 mais ce n'est pas le sujet ici, il faut être informaticien avec de l’expérience et de la logique pour trouver une solution de manière autonome.
> 
> 
> 
> Il est inutile d'appeler Free concernant le changement d'une IP, surtout quand elle est locale et que tu peux paramétrer toi même comme tu le souhaite cette IP depuis l'interface de ta freebox via ton navigateur, ton téléphone via l'application spécifique de Free, et même depuis l'interface d'administration disponible sur le site de free.fr
> 
> 
> 
> Par défaut il n'y a pas d'IP renseigné dans "Passerelle IPV4 et Diffusion IPV4" de l'interface LAN, puisque c'est lui même le routeur.
> 
> 
> 
> Sans OMR, tu peux parfaitement accéder à l'interface de ta freebox en paramétrant une IP fixe sur ton ordinateur dans la même plage IP que ta freebox.
> 
> 
> 
> Le projet OMR est encore en phase BETA, de mémoire il n'y a qu'un seul développeur, pour ma part j'apporte mon soutiens comme je le peux afin de debug et soutenir un projet prometteur.
> 
> 
> 
> dans tes 17 messages en moins de 3 heures, afin de trouver une solution rapidement, je te suggère de : 
> 
> 
> 
> 1°) Répondre à la demande de Ysurac : "Pour ça il me faudrait la config de la partie network. Donc dans Système->OpenMPTCProuter le dernier onglet (juste la partie network) ou via SSH ou Services->Terminal uci show network"
> 
> 
> 
> 2°) retourner dans une ancienne version qui fonctionnait avec ton matériel, restaurer une sauvegarde fonctionnel, et si c'est OK je te suggère fortement de ne surtout plus toucher à rien.
> 
> 
> 
> Je suis désolé de cette intervention, mais ce n'est pas la première fois que j'ai remarqué tes multiples messages en mode SPAM et que je trouve personnellement désagréable à lire.
> 
> 
> 
> Avec plaisir de t'apporter des réponses techniques logiques, et bonne fête de fin d'année.

Merci pour ta réponse.

En fait, j'ai appelé Free pour être sur que l'IP interne de la Freebox n'avait pas changé. Et c'est le cas. Pourtant, impossible d'y accéder sur Chrome. Je ne comprends pas pourquoi. Ils m'ont juste confirmé que l'ip est toujours 192.168.0.1

J'ai tenté plusieurs fois de revenir sur des sauvegardes mais toujours l'erreur Luci main.

Mais la, vu que OMR fait bien routeur mais n'a accès ni à la Freebox ni au routeur 4G, j'ai l'impression qu'il y a quelque chose sur le réseau LAN qui bloque tout.
Pourtant, jusqu'à cet après midi tout marchait parfaitement. C'est en redémarrant OMR en début d'après midi que tout est parti en sucette.

Désolé pour les messages mais j'ai une maison connectée avec plus de 120 appareils wifi et la plus rien ne marche. Mes Jeedom, Home assistant...
J'ai perdu toute la configuration DHCP et by-pass.

J'essaierai demain de régler le problème Luci et je vais tout éteindre.

C'est possible qu'un appareil sur le réseau foute le bordel?

JeedHome44, 2021-01-01:
Bonjour,

Je viens donc de remettre proprement la 0.57.2 et ma sauvegarde d'hier après midi. Pour ouvrir l'interface web je dois faire /sbin/rpcd en ssh mais je n'ai pas compris comment la rétablir sans passer par ssh. est-ce possible de me le dire? Merci

JeedHome44, 2021-01-01:
> Pour ça il me faudrait la config de la partie network. Donc dans Système->OpenMPTCProuter le dernier onglet (juste la partie network) ou via SSH ou Services->Terminal `uci show network`

Voici la partie demandé:

`network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='1'
network.globals=globals
network.globals.ula_prefix='fd70:dd6a:1e0a::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_scheduler='default'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='dc:a6:32:7a:49:37'
network.lan.metric='2'
network.lan.label='LAN OMR'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.addlatency='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.defaultroute='0'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.macaddr='e6:af:f0:8a:ed:7e'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.0.1'
network.wan1.label='Freebox'
network.wan1.ipaddr='192.168.0.2'
network.wan1.multipath='master'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.defaultroute='0'
network.wan2.ifname='wan2'
network.wan2.masterintf='eth0'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.macaddr='da:fb:b0:5b:13:bf'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.8.1'
network.wan2.label='Huawei'
network.wan2.ipaddr='192.168.8.2'
network.wan2.metric='4'
network.wan2.multipath='on'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='1000'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omrvpn.metric='1200'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.auto='0'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'
network.omr6in4.ipaddr='10.255.251.2'
network.omr6in4.peeraddr='10.255.251.1'
network.omr6in4.metric='1201'`

JeedHome44, 2021-01-01:
Mais ce que je trouve vraiment étrange c'est qu'un simple redémarrage de OMR rend complètement impossible maintenant l'accès à mes 2 clients web sur mon LAN. Que ce soit la freebox en 192.168.0.1 ou la box 4G en 192.168.8.1, je ne peux pas y accéder et OMR non plus... Pourtant, avec l'application Freebox sur iPhone, si je me mets en 4G, j'ai bien accès à l'interface mobile.
Est-ce possible qu'en redémarrant OMR, un des appareils de mon réseau bloque quelque chose? J'ai 2 routeurs TP link à 2 endroits de la maison réglé en mode Access point et 2 répéteur sur ethernet CPL réglé aussi en Access Point. Seul OMR route...

Ysurac, 2021-01-01:
Pour l'erreur avec luci, via SSH obligatoirement (vu que c'est le seul moyen d'accès dans ce cas):
```
uci -q set rpcd.@rpcd[0].socket=/var/run/ubus/ubus.sock
uci -q commit
```
Sinon je ne comprend pas ta configuration réseau, wan1 et wan2 utilisent eth0 en interface mais elles ne sont pas définis comme MacVLAN, ce qui devrait être le cas.
C'est normal quand le routeur, ce qu'est OMR, ne fonctionne pas (ou pas correctement) que tu ne puisses plus accéder à la Freebox et box 4G qui sont sur d'autres réseaux. C'est le rôle du routeur de justement permettre de trouver la bonne route et diriger vers ces réseaux.

JeedHome44, 2021-01-01:
Bonjour,

J'ai toujours mis en normal et ça marchait très bien.

Pourquoi après un reboot ca ne marche plus en Normal?

Je viens de tester de mettre sur MacVLAN et tout est rentré dans l'ordre.

J'ai du louper un truc... mais j'aimerai bien comprendre 😄 

Tu viens de m'enlever une énorme épine du pieds merci!

Toute cette galère hier que pour ça... curieux que ça marchait bien avant.

Ysurac, 2021-01-01:
Certainement car il y avait des restes de la configuration MacVLAN. En normal ça ne peut pas fonctionner correctement quand on a un seul port, sauf via l'utilisation de VLAN.

JeedHome44, 2021-01-01:
Encore merci pour ta réponse si tôt.
Mon problème alors de déconnexion de la 4G régulièrement venait donc sûrement de la aussi... ce n'était pas mon routeur ni le débit le fautif mais la configuration sur Normal

closed 2021-04-06 by github-actions[bot]