Ysurac/openmptcprouter #1490: Question: Traffic shaping with SQM

phantomFaceTime, 2020-12-31:
Question: I have a user that I would like to stop using too much bandwidth.  Normally on OpenWRT I think I could do this with SQM QoS using cake/piece of cake.  But it looks like SQM would not work well with openmptc.  I tested by applying it to each wan link and it slows the overall connection down significantly.  Is that reasonable for mptc?  What would be a good way on openmptcrouter to constrain a single user/mac address/ip address from using too much download bandwidth?  Thank you very much for any hints.

Ysurac, 2021-01-04:
Maybe installing something like ```luci-app-nft-qos``` could help you.

phantomFaceTime, 2021-01-05:
Thank you very much for the suggestion: I have now tried installing luci-app-nft-qos and it does look like it should do what I need. However, it also appears to kill the dhcp server so obviously I need to investigate further.

And thank you also for developing this wonderful software -- it really does work wonders.

closed 2021-04-11 by github-actions[bot]