Ysurac/openmptcprouter #1493: Linksys WRT32X, cannot access 192.168.100.1 from the WAN port & Not supported warning on sysupgrade image on LuCI

arinc9, 2021-01-01:
## Expected Behavior

To be able to access the router at 192.168.100.1 from the WAN port.

## Current Behavior

Cannot access LuCI after flashing the sysupgrade image.
LuCI puts up a warning:
[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. Download and install sysupgrade image from LuCI.
  2. Connect to the WAN port
  3. Change subnet on PC to 192.168.100.0/24
  4. Try connecting to 192.168.100.1

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.57.2 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: Linksys WRT32X

Ysurac, 2021-01-01:
Yes, as it's specified in download page and in the text before, config can't be migrated from version < 0.55

arinc9, 2021-01-01:
I’ve made sure to not keep the current config. I’m trying to update from 19.07.4. Am I doing something wrong?

Ysurac, 2021-01-01:
Update from OpenWrt are not tested/supported/no idea.

arinc9, 2021-01-01:
Oh, ok. Thanks for the heads up.

arinc9, 2021-01-01:
Reopening this issue because OEM firmware won't accept the image.
Here's OpenMPTCPRouter image, which the firmware rejects:
[image]

Here's OpenWRT image which firmware accepts:
[image]

WRT32X is V1.0. The OpenMPTCPRouter image is V1.1. Are you sure this is the correct firmware for this device?

Ysurac, 2021-01-01:
Yes I'm sure, but possible that the image take too much space for WRT32X and that I need to remove some drivers/softwares.

arinc9, 2021-01-02:
I've diagnosed the firmware on serial connection. It boots fine. Here's how much the image takes on the flash.

```
root@OpenMPTCProuter:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                37.5M     37.5M         0 100% /rom
devtmpfs                248.7M         0    248.7M   0% /rom/dev
tmpfs                   249.2M    148.0K    249.1M   0% /tmp
/dev/ubi0_1              66.1M    356.0K     62.3M   1% /overlay
overlayfs:/overlay       66.1M    356.0K     62.3M   1% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

Looks like the switch won't work on the firmware hence why I can't reach the router.
```
root@OpenMPTCProuter:~# swconfig list
(nothing)

root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 60:38:E0:C9:0C:10
          inet6 addr: fe80::6238:e0ff:fec9:c10/64 Scope:Link
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1508  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:37

lan1      Link encap:Ethernet  HWaddr 90:28:35:21:62:B8
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan2      Link encap:Ethernet  HWaddr B0:27:9D:98:D8:BC
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan3      Link encap:Ethernet  HWaddr BA:6D:56:95:6D:85
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lan4      Link encap:Ethernet  HWaddr 32:C7:65:34:33:62
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1299 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1299 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:85529 (83.5 KiB)  TX bytes:85529 (83.5 KiB)

wan       Link encap:Ethernet  HWaddr 60:38:E0:C9:0C:10
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
[... 86 lines cut ...]
        option ipv6 '0'
        option metric '12'

config interface 'omrvpn'
        option ifname 'tun0'
        option proto 'dhcp'
        option ip4table 'vpn'
        option multipath 'off'
        option leasetime '12h'
        option type 'tunnel'
        option txqueuelen '100'
        option ipv6 '0'
        option metric '1200'

config interface 'omr6in4'
        option proto '6in4'
        option ip4table 'vpn'
        option multipath 'off'
        option ipaddr '10.255.255.2'
        option peeraddr '10.255.255.1'
        option gateway 'fe80::a00:1'
        option ip6addr 'fe80::a00:2/128'
        option auto '0'
        option metric '1201'
```

Ysurac, 2021-01-02:
WAN interface seems to be correctly set. I don't see real error here.

arinc9, 2021-01-02:
There’s no configuration about the switch on `/etc/config/network`. `swconfig` doesn’t detect a switch. Maybe you have not included the driver for the switch in this build?

Ysurac, 2021-01-02:
DSA is used instead of swconfig now.

arinc9, 2021-01-02:
OK, let's just forget about it.

closed 2021-01-02 by arinc9