Ysurac/openmptcprouter #1494: A lot of daemon.err /usr/bin/ss-redir

dettofatto, 2021-01-01:
i have a lot of
 daemon.err /usr/bin/ss-redir remote recv: Connection reset by peer  
I have  this issue from 0.55, but also now with 0.57.2
And 5.4.81-mptcp 0.1023 on my raspberry  pi4.
During this issue  i can't browse.

Ysurac, 2021-01-01:
Please use the template !
This error is related to connection problem.

dettofatto, 2021-01-08:
I'm sorry. 

## Expected Behavior

Stable connection 

## Current Behavior

In my system log, from version 0.54 to 0.57 I have a lot of "micro" disconnections, that I can see also browsing... 
I read in system log a lot of
daemon.err /usr/bin/ss-redir remote recv: Connection reset by peer
And similar errors. 

## Possible Solution

Really don't know. I've disconnected openmptcprouter and connected my mikrotik (setting it balanced) for seeing if there are these disconnections from my 2 isp(2 4g+ with Huawei modem), but I didn't see any issue. 

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

May be, my vps is faulty? How can I check? 
I have the same vps from 0.44, so, now.?

## Specifications

  - OpenMPTCProuter version:  0.57.3
  - OpenMPTCProuter VPS version: 
  - 0.1024
  - OpenMPTCProuter platform:  rpi4 64 bit

closed 2021-04-14 by github-actions[bot]