Ysurac/openmptcprouter #1496: Only one connexion is used

Nethary, 2021-01-01:
Hi, i have an issue with my OpenMPTCP conf.
## Expected Behavior
I want to have my two connexion used.

## Current Behavior
Only my 4G connexion is used. I don't have the VPS public IP, always my WAN 2 public ip. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version:  0.1017
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

here is the router conf : 
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/jeuxvideo.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/static.jvc.gg/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/gcsip.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/secure.square-enix.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/pass.canal-plus.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/decouvrir.mycanal.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/www.mycanal.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/e8933.b.akamaiedge.net/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/reddit.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/go.ocs.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/ocs.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/netflix.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/bigfernand.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/eu-west419.discord.gg/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/crunchyroll.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/cloudflare.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/mafreebox.freebox.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/store.steampowered.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/particulier.edf.fr/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/tpeweb.paybox.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/paybox.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/google.com/omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/instant-gaming.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/vod.canalplus.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/ecoledirecte.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/eurofins.mesanalyses.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/intermarche.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/itmconnect.mousquetaires.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/drive.intermarche.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/mynintendostore.nintendo.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nintendo.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/https://www.dailymotion.com//omr_dst_bypass_wan2,omr6_dst_bypass_wan2' '/api.accounts.nintendo.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/cdn03.nintendo-europe.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1,omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.net/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxso.net/omr_dst_bypass_all,omr6_dst_bypass_all,omr_dst_bypass_all,omr6_dst_bypass_all' '/nintendo.net/omr_dst_bypass_wan2,omr6_dst_bypass_wan2,omr_dst_bypass_wan2,omr6_dst_bypass_wan2'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_default='0'
dhcp.lan.dhcpv6='disabled'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
[... 1464 lines cut ...]
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

Ysurac, 2021-01-02:
What do you have in status page ?

Nethary, 2021-01-02:
Everythings looks good 
[image]

Ysurac, 2021-01-02:
How do you check that only one connection is used ?

Ysurac, 2021-01-02:
And you should have VPS IP as output in your case. Check that your computer use OpenMPTCProuter DHCP or OpenMPTCProuter IP as gateway.

Nethary, 2021-01-02:
I check with the trafic page when i do speedtest.
I think i have found my problem. That was some entry on my OMR bypass configuration... 
I have made menage on certains URL (some with http:// at the beginning or / in the end) and that's seems to be good since. 
I'll try to reproduce that when i have time if you want.

closed 2021-04-07 by github-actions[bot]