Ysurac/openmptcprouter #1497: Free Mobile ISP 4G +new ipv6 option => OMR cant aggregate

vinceducat, 2021-01-02:
## Expected Behavior
i aggregate adsl and 4G from Free and Free Mobile.
aggregate works even ipv6 is active...

## Current Behavior
when ipv6 option is activate for the 4g link in my free mobile account
the aggregation doesnt work

## Possible Solution
only autorise ipv4 in my B715 router ..

## Steps to Reproduce the Problem

  1. activated ipv6 optin on free mobile account
  2.
  3.

## Context (Environment)
Free Mobile for 4G 
Free For adsl
Milkywan for vps

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 0.1023
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-01-02:
I think your computer get an IPv6 from free mobile, so all IPv6 traffic use this.

vinceducat, 2021-01-03:
So the best is to leave ipv6 disable in the B715 router ?

Ysurac, 2021-01-03:
yes

closed 2021-01-03 by vinceducat