Ysurac/openmptcprouter #1500: Bandwidth calculations

surikat, 2021-01-04:
Hello and happy new year 2021 !

Is there an approved note for this system that explains theoretical calculations of aggregate bandwidth based on devices, number of interfaces, etc.?
Is total bandwidth really equal to the sums of each interface bandwith or it's fully utopian? For instance, is it possible to reach 300 Mbits/s with two 4G routers (150 Mbits/s each).
 
Which elements could be responsible for restricting bandwidth ?

Sorry for these many questions but this system is very interresting and thanks to makers/helpers

Ysurac, 2021-01-04:
We can't get 100% of total bandwidth for a single TCP stream. Some tests give about 86% of full speed with 6 connections for a single stream. We get a better bandwidth usage with multiple streams.
But some bandwidth is used by encryption and 4G/5G/Wifi connections are losing many packets and with a not so low latency, so aggregation result can't be exactly 100% bandwidth aggregation.
VPS and router platform can (and it's often the case) restrict possible speed due to slow CPU or not enough memory.

surikat, 2021-01-04:
Thank you for your answer. It is quiet difficult to understand this complex system and it seems finally not easy to master it, because a lot of parameters and factors, and surely OpenMptcpRouter is in experimental state.. When you work in computer science, you do not work (or rarely) for random results, because computer science is based on binary. So everyone wanting to move to OpenMptcprouter should know that the end result cannot be anticipated accuratly and will be variable (no guarantee). In my case, I do not want just have a better connection that with ADSL. I would love reach the best bandwith as possible with 2,3, or 4 interfaces, good VPS, good raspberry, good switch, good ethernet cables,...

Ysurac, 2021-01-04:
With a random connection and a random configuration you will always get a random result. It's always the case in computer science.

closed 2021-01-04 by Ysurac

surikat, 2021-01-04:
why random configuration, for who, for me? Wireless connections are not random. Precisely, computing science and source coding help random elements to be stable / accurate.

Ysurac, 2021-01-05:
wireless connections are not stable and it's even less stable for 4G/5G connections. Any wireless connection have more packet loss than a fiber connection. On a 4G connection speed is often not stable. Nothing specific to OpenMPTCProuter here. A bad connection will give a bad result, to make it stable you will always get same thing: losing speed.

For the hardware, it's always same problem, you can't get 10Gb/s on a RPI. There is some issues opened for hardware tests, people that want a high speed connection use some x86_64 based hardware with AES-NI and a good VPS that have enough CPU/memory/bandwidth.

Nobody can say that you will get always full network speed of all wireless connections on any existing hardware. Not possible. Same hardware with same connections (at same time/day for wireless connections) will always give same result, as any router.

surikat, 2021-01-05:
Indeed. Thank you for the time spent explaining and working on this great project. Can you give me example links about MPTCP speed testing please? MPTCP congestion is apparently important to improve throughput. Is there a better algorithm (BALIA, I read) or it depends on the context, and which congestion algorithm was chosen for OpemMptcpRouter? I will share my feedback with you for my future tests if you are interested.

Ysurac, 2021-01-05:
It depends on context. BBR give good result with good connection but Cubic give a far better result on bad connections.
I added BBRv2 alpha that seems to give some good result in more case than BBRv1 but it still need more testing.
I never have a really good result with olia, lia,... tcp congestion algo, but they can give a little better result than cubic in particular case (that I don't have with my own connections and no testers seems to have good result with them too).

For MPTCP speed testing there is some tests on research articles but often in lab conditions. OpenMPTCProuter have some optimizations and it use latest experimental out of tree multipath TCP version. Some user have more than 1Gb/s but it's not something I can test and optimize with my connections for now.