Ysurac/openmptcprouter #1501: Unable to access GUI with 0.57.2 on ESXI

kams19, 2021-01-04:
HI,
Happy New Year to everyone.
I am currently having v0.56.5 on ESXi, and keep getting update prompts. I took a plunge and downloaded the new OVA file -
Open Virtual Appliance image (EFI): openmptcprouter-v0.57.2-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi.ova (2020/12/31 - 71.83Mo - sha256sum)
Since then I am not able to access the GUI.  I have verified the interfaces are connected to correct vswitch, etc. 
Is that happening just to me or is it a common problem? I have reverted back to 0.56.5 and that works without any issue.

Ysurac, 2021-01-04:
Why are you using EFI image on ESXI ?
What do you call update prompts ?

kams19, 2021-01-04:
i believe I had used the efi image for the previous version as well, hence used the same now
Sorry didnt get your second question.

Ysurac, 2021-01-04:
Try using not EFI image and tell me what you have in ESXi console/KVM.

kams19, 2021-01-04:
Looks like that was the problem.. dont know how it worked with EFI image in the earlier version.
One more feedback, the Glory TUN TCP comes up superfast in this version. In earlier version it used to take 30min -1hr

closed 2021-04-10 by github-actions[bot]