Ysurac/openmptcprouter #1504: IPV4/IPV6 DSVPN

jfh75, 2021-01-05:
Hello,

I have noticed a strange ipv6 address for DSVPN. Also, when using DSVPN, it is not working anymore.
Why is it I get a unvalid ipv6 for DSVPN ? Is there a way I can force IPV4 ?
Here is the welcome screen when logging to VPS
System load:                  0.09
  Usage of /:                   20.7% of 54.27GB
  Memory usage:                 3%
  Swap usage:                   0%
  Temperature:                  36.0 C
  Processes:                    135
  Users logged in:              0
  IPv4 address for dsvpn0:      10.255.251.1
  IPv6 address for dsvpn0:      64:ff9b::aff:fb01
  IPv4 address for enp1s0:      192.168.2.5
  IPv6 address for enp1s0:      2a01:e35: etc... etc....
  IPv4 address for gt-tun0:     10.255.255.1
  IPv4 address for gt-udp-tun0: 10.255.254.1
  IPv4 address for mlvpn0:      10.255.253.1
  IPv4 address for omr-bonding: 10.255.248.1
  IPv4 address for tun0:        10.255.252.1
  IPv4 address for tun1:        10.255.250.1

Also, when running update screen, I first got a warning that some other process prevented from upgrading (is another process using it var/frontend etc.
Relaunching the script went fine afterwards
Connection to OMR is lost since though. I guess I have to reboot OMR ?

Thank you for your help,

JFH

VPS : 1023 on ubuntu 20.04
OMR 56.5 x86_64

jfh75, 2021-01-05:
I have tried to update my old debian VPS install on another SSD, everything went fine.
But I noticed with "ipaddr" that there are plenty of OVPNBONDING devices I don't have on the ubuntu install ?!
Is that normal ? Or precisely a difference between VPS 1022 and 1023 ?

Thank you for your help,

JFH

jfh75, 2021-01-06:
Hello,

After upgrade to OMR 57.3, it works fine again with 1023 debian based VPS.
You definitely should put only debian as supported on VPS. And drop ubuntu, it is obviously not working.
It is the second time it doesn't work with ubuntu for me. If you remember upgrade to 1022 went with two python things missing for no reason with no warning.
Thank you
JFH

Ysurac, 2021-01-06:
Recommended install is on Debian as indicated in the documentation. Ubuntu install is working in most case but some strange install may cause issues.

jfh75, 2021-01-06:
Yes, I don't want to explore as it is working with my debian VPS, but it seems it only works well on a fresh clean install. Like last time I retried when you put ubuntu server 20.04.
Thank you !
JFH

jfh75, 2021-01-06:
And now I have download peeks back at 1.9Mo/s with 10MB adsl and B818 Huawei free 4g :=))
JFH

closed 2021-04-12 by github-actions[bot]