Ysurac/openmptcprouter #1505: Question: Aggregate more than 8 connections

BastiGER, 2021-01-05:
Hi! 

Is there a possibility to connect several routers in series, that you can aggregate e.g. with two OpenMPTCPRouters 16 instead of 8 connections? 

best regards

Ysurac, 2021-01-05:
Maybe, but I'm not sure you will get a really good result.
Should also be possible to patch Multipath TCP to support more connections but I don't have enough connections for testing that.

surikat, 2021-01-05:
Hi. Already, is there a significant speed gain between 2, 4, 8 good connections ? Maybe there is a limit from which it is unnecessary to add new interfaces. For home use, it is little crazy, expensive and difficult to use more that 4 connexions.

Ysurac, 2021-01-05:
There is a significant speed gain and this add some failover.
And more speed we have, more speed we use ;)

Malaga82, 2021-01-05:
i started myself with 2 and now i'm with 3, and from 2 to 3 i found "complications" skyrocketed, i can't imagine deal with 8 connections!

surikat, 2021-01-05:
For your two LTE connections, do you have the same LTE router brand/model + same antenna model, and also the same ISP, same plan?  Many conditions can make differences.

Ysurac, 2021-01-05:
@Malaga82 I don't see any open issue, if you have problems open one.
Keep this issue subject on initial message please.

Malaga82, 2021-01-06:
@surikat i have different LTE router and provider. Having same 4G provider for more than 1 wan is not optimal, since i will hit the same BTS and core network.

@Ysurac it wasn't mean to override the discussion with my "problems", just to inform him that deal wih many wans could be a lot of work, and maybe even more if wans are LTE based.

surikat, 2021-01-07:
@Malaga82 I have tested 2 configurations. One with 2 4G routers connected to the same provider A.  And a second 
conf. with 2 different providers A and B, but A provides a better throughput than B. Well, I have significant better ping/download/upload with the same provider configuration. No problem with using the same BTS.

linuzel, 2021-01-09:
I tested 7 connections on the same isp (dsl) and it does work, it is also very demanding.
On an apu3 I can’t get the full bandwidth due to very high cpu usage.

It get better without shadowsock encryption but omr-tracker use more than 25% cpu idling.

it seems to be fine with un to 4.

surikat, 2021-01-09:
@linuzel  Why don't you change apu3 to Raspberry PI4B? Actually, what's the best gear for OpenMptcpRouter?

Adorfer, 2021-01-09:
> @linuzel Why don't you change apu3 to Raspberry PI4B? Actually, what's the best gear for OpenMptcpRouter?

Not so good joke. (and a derail)

Back to topic: If you are going on so many links, you will probably need real core-i and equivalent current AMD processors. And perhaps a LAN capable of 2.5GBit/s or more, so likely some SFP+ with DAC to the switch. 

And do not forget that the load is on both sides, OMR and VPS. Make sure that you have enough (dedicated, not overprovisioned CPUs cores) 

But do i understand correct, there is currently a hard limit on e.g. 8 WAN interfaces? (initial issue did not use template, so it's a bit hard to guess what the problem is.)

Ysurac, 2021-01-09:
Multipath TCP code limit to 8 WAN interfaces, I can patch it to support 16 or more, but I'm not able to test if this will work correctly, so not sure if it's a good idea ;)

linuzel, 2021-01-09:
> Multipath TCP code limit to 8 WAN interfaces, I can patch it to support 16 or more, but I'm not able to test if this will work correctly, so not sure if it's a good idea ;)

That was the point of my message actually.
Even patched with 16 WANs it would require a very powerful device to support decent speed.
In my example I can’t manage 6x8Mbps on an APU (which is a pretty capable device).

I hope to have the occasion to test it on a stronger system, it might be interesting

closed 2021-04-14 by github-actions[bot]