Ysurac/openmptcprouter #1506: Not possible to make the update from 0.57.2 to 0.57.3

xillibit, 2021-01-05:
## Expected Behavior
When doing the update from 0.57.2  to 0.57.3 you shoudn't have issues

## Current Behavior
After the image of 0.57.3 is uploaded, it shows some errors : 

[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. On OMR goes into system, then backup /update
  2. Click on select the image to write
  3. When the image is uploaded at 100% you get the errors

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.57.2
  - OpenMPTCProuter VPS version: 0.1023
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-01-05:
Check that FS is not read only and make sure image is the correct one.

Malaga82, 2021-01-06:
@xillibit i have promox too and updated fine.
i used this image: openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined.img.gz
i have qcow2 as disk format, if can help you.

If you have issue, i suggest to make a config backup, download the image ( virtualbox ), convert it to qcow2, upload it changing name etc ( you should now ) and then import the config saved.

xillibit, 2021-01-06:
@Ysurac : 

I don't know how to check if the filesystem is on read-only, i use this image : openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined.img.gz . I double check the name of image downloaded because i already one the mistake before to import the wrong and to break the install

@Malaga82 : when i have created the vm, i have converted the image of openmptcprouter in raw format. How do-you convert the image in qcow2 format ?

Ysurac, 2021-01-06:
@xillibit Image is the correct one. Check the result of ```mount``` via SSH or Services->Terminal on the router and check what you can have in State->Kernel log.
Check also that host disks are not full or read only.

Malaga82, 2021-01-06:
@xillibit .\qemu-img.exe convert -f vdi -O qcow2 file.vdi file.qcow2
you can fine qemu-img here: https://cloudbase.it/qemu-img-windows/

xillibit, 2021-02-13:
@Ysurac : i double-check the image before to do the update. When i go into Services->Terminal i have that : 

[image]

What i need to looking for into the kernel log ?

`root@OpenMPTCProuter:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               495.9M    211.0M    274.7M  43% /
devtmpfs                512.0K         0    512.0K   0% /dev
tmpfs                   242.3M    892.0K    241.4M   0% /tmp
/dev/sda1                63.0M      5.0M     56.7M   8% /boot
/dev/sda1                63.0M      5.0M     56.7M   8% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
`

@Malaga82 : thanks for the tips

Ysurac, 2021-02-14:
For Services->Terminal, it's because "openmptcprouter" name doesn't resolve (should be if OMR DNS are used).
And I asked ```mount``` not df ;)
For kernel log check any information about I/O errors.

xillibit, 2021-02-14:
> root@OpenMPTCProuter:~# mount
/dev/root on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=245984k,nr_inodes=61496,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda1 on /boot type ext4 (rw,noatime)
/dev/sda1 on /boot type ext4 (rw,noatime)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
mountd(pid7031) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=1,timeout=21474836510,minproto=5,maxproto=5,indirect)

Not found errors about I/O in kernel log, i have pasted it here : https://gist.github.com/xillibit/3e15fc8c80adfe9be52f53da7832f15a

Ysurac, 2021-02-14:
Nor problems here.
Check if there is no issue on host, and try again to update.

xillibit, 2021-02-15:
I think i figured out, when i imported the image into my VM on proxmox it was done on wrong place and in this place there is not enough disk space

closed 2021-02-15 by xillibit