Ysurac/openmptcprouter #1507: WiFi-Driver: AMPAK AP6212 / Broadcom BCM43430

ByteArrayException, 2021-01-05:
Hey!
I use the Broadcom BCM43430 (AMPAK AP6212) in my x64 machine but it doesn't recognized by openmptcprouter.
Any ideas?

Best regards

Ysurac, 2021-01-05:
You may need to install brcmfmac-firmware-43430-sdio package.

ByteArrayException, 2021-01-05:
Sorry (as I said in one of the last issues) I am not very familiar with OpenWRT, do you have an example command? 
Thanks in advance

Ysurac, 2021-01-05:
In System->Software, use "update list" and search package using "filter" then install it.

ByteArrayException, 2021-01-05:
Thanks! I will test that tomorrow and leave an reply.

ByteArrayException, 2021-01-16:
Hey,

i tested this driver, the "kmod-brcmfmac" and the "brcmfmac-firmware-43430a0-sdio" Driver but none works. I also tested an Wifi-Stick with Ralink RT5572 Chipset. I installed the Drivers (it was already installed) but it would not recognized.

Any idea?

Thanks in advance

best regards

Ysurac, 2021-01-17:
You can check on OpenWrt forum if you find something.
And check State->Kernel log if it's really not detected.

ByteArrayException, 2021-01-17:
Hey,

The log only shows that the kmod modules are registering or initialising. However, (even if I only plug the USB stick in or out and in after the boot) there is no entry with a note that a USB stick is initialised by Ralink, on a Windows computer it worked straight away.
For the AMPAK i found yesterday an workaround, i will test this in the next days and give here an reply.

Best regards

ByteArrayException, 2021-03-06:
Hey,

now it working fine. But: The radios detected in UI under Network -> WLAN, but constantly "not active" and scan is also not possible. Have you any workaround for this?

Best regards

closed 2021-03-07 by ByteArrayException