Ysurac/openmptcprouter #1508: RPI4 64bits usb to eth dongles make the file system read-only when 2 or more are connected.

gartz, 2021-01-06:
## Expected Behavior
When connect 2 or more USB to ETH dongles in a RPI4 64bits, it should let use eth1, eth2, and so on...

## Current Behavior
Once the second USB to ETH dongle is connected, the filesystem switch from read/write to read only.
After that it's impossible to apply any configuration, all the configurations stay in the unsaved changes.
The eth from every connected dongle is detected and displayed dispite of the bug.

## Possible Solution
I honestly don't know why this happens or how to fix it.
The only way I can fix when that happen is to flashing again the SD card.

## Steps to Reproduce the Problem

  1. connect one USB to ETH dongle in the RPI4
  2. add the interface and save+apply
  3. connect the second USB to ETH dongle in the RPI4
  4. add the interface and save+apply (it will show as unsaved, now the filesystem became read only)

## Context (Environment)
I have network that uses DHCP from my provider, so using a switch is not working, as the DHCP from the provider messup with IPs and don't let the LAN get the IP provided by the Openmptcprouter DHCP server.
When using separated interfaces, that problem is no more. Therefore I need to use 2 USB to ETH dongles, one for each provider. and the ETH0 to my local network. Unfortunately that is impossible due to the error that only let me do with one dongle.

## Specifications

  - OpenMPTCProuter version: 0.57.1, 0.57.2, and 0.57.3
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: RPI4_64

Ysurac, 2021-01-06:
Check State->Kernel log
But I would say it's using too much power for your power supply or for the RPI4.

henry18hn, 2021-02-17:
Same setup, yesterday no problems, today, read only problem . Read only problem started when I had to re enable DHCP on 1 LTE Openwrt router/modem to regain connection because my providers IP changed and MPTCRouter was not able to receive new IP

henry18hn, 2021-02-17:
I have some 5a I use for my openwrt devices, Ill give that a try. I looked through Kernel logs, I have not found State yet, only these:

[   34.572084] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sh: deleted inode referenced: 983
[  157.179595] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sed: deleted inode referenced: 983
[  206.425357] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sed: deleted inode referenced: 983
[  215.726399] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sh: deleted inode referenced: 983
[  223.136711] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sh: deleted inode referenced: 983
[  272.488271] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sed: deleted inode referenced: 983
[  289.217781] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sh: deleted inode referenced: 983
[  308.789476] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sed: deleted inode referenced: 983
[  316.337625] EXT4-fs (mmcblk0p2): error count since last fsck: 137
[  316.344841] EXT4-fs (mmcblk0p2): initial error at time 8: ext4_lookup:1705: inode 956
[  316.353712] EXT4-fs (mmcblk0p2): last error at time 1613578376: ext4_lookup:1705: inode 956
[  328.133166] EXT4-fs error (device mmcblk0p2): ext4_lookup:1705: inode #956: comm sed: deleted inode referenced: 983

Ysurac, 2021-02-17:
In System->Mounts points, you can check "Check filesystems before mount", but if the FS is corrupted during run time it's often a faulty power or a bad SD card.

Adorfer, 2021-05-18:
When connecting more than 1 USB-NIC: Use a **powered** USB Hub! otherwise there is multiple risks of instability. 
(damaged FS on the SD is just one. lost eth-link, or even crashes/reboot during heavy load is not acceptable.)

Bumpalert, 2021-05-19:
Good Advice. I already suspected that and ordered a usb hub

Bumpalert, 2021-06-02:
It's simple to repair.
First connect to shell (via ssh).
Type:
fsck.ext4 -y /dev/mmcblk0p2
reboot

on next start your system will be repair and ready to write.

as suggested by someone to me in another post.

closed 2021-09-06 by github-actions[bot]

spacesuitdiver, 2021-10-18:
@gartz did you get your setup to stabilize? I'm not even plugging any USB into my rpi4 and it still boots up as read-only.