Ysurac/openmptcprouter #1510: Problème firewall depuis la version 0.57.3

kdesroches89, 2021-01-06:
Bonjour,
Depuis la mise à jour en 0.57.3 j'ai un soucis firewall , les services qui sont normalement en sortie sur le VPS ne répondent plus et il semble que ça soit un soucis lié au firewall . Actuellement par exemple : https://maison.kikimath.net indique "que le délai d'attente est dépassé"

Merci par avance pour l'aide .

Voici les logs système : 

Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty39, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice firewall: Reloading firewall due to ifup of wan2 (wan2)
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty29, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty29
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty29, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty57, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty57
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty57, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty19, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty19
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty19, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty2, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty2
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty2, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty47, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty47
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty47, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty37, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty37
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty37, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty27, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty27
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty27, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty55, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty55
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty55, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty17, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty17
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty17, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty0, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty0
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty0, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty45, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty45
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty45, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty35, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty35
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty35, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice unbound: default memory configuration
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty63, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty63
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty63, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty25, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty25
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty25, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty9, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty9
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty9, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty53, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty53
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty53, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty15, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty15
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty15, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty43, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty43
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty43, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty33, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty33
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty33, subsystem=tty
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty61, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty61
Wed Jan  6 09:33:36 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty61, subsystem=tty
[... 517 lines cut ...]
Wed Jan  6 21:51:42 2021 daemon.info dnsmasq-dhcp[26544]: DHCPACK(eth0) 192.168.100.194 68:a3:78:52:a4:26
Wed Jan  6 21:51:42 2021 daemon.warn dnsmasq-dhcp[26544]: not giving name Mathilde-tel.kikimath.net to the DHCP lease of 192.168.100.117 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.226
[... cut ...]
Wed Jan  6 22:37:13 2021 daemon.warn dnsmasq-dhcp[26544]: not giving name kikimath.kikimath.net to the DHCP lease of 192.168.100.198 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 169.254.115.125
Wed Jan  6 22:40:33 2021 daemon.err /usr/bin/ss-local[22178]: getpeername: Socket not connected
Wed Jan  6 22:49:01 2021 user.notice post-tracking-post-tracking: Check API configuration...
Wed Jan  6 22:49:04 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Wed Jan  6 23:09:31 2021 daemon.err uhttpd[3984]: luci: accepted login on / for root from 192.168.100.127

Ysurac, 2021-01-07:
Il n'y a pas de changement au niveau du firewall dans cette version. Il faut essayer de revalider le wizard dans Système->OpenMPTCProuter afin que les informations soient mise à jour côté serveur et sinon https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

kdesroches89, 2021-01-08:
Merci @Ysurac

closed 2021-01-08 by kdesroches89