Ysurac/openmptcprouter #1511: Wi-Fi client mode for OMR?

topbilling, 2021-01-07:
I'm using a Raspberry Pi 4 as the MPTCP router.  Can I setup the Pi to connect to a local WLAN in client mode and aggregate that connection along with the others I'm using?

Ysurac, 2021-01-07:
Yes, but RPI wifi is not really good.

topbilling, 2021-01-07:
Yes, I read that it's best to only use 2.4 Ghz.  What are the steps to enable this?  I added the WLAN interface, but it the status page shows gateway down even though the connection is showing ongoing data received and transmitted.

topbilling, 2021-01-07:
I should have been more specific.  I know how to enable the WiFi client using 2.4 Ghz and add it as an interface, but it shows a "Network device not present message" on the main interfaces page even though data is being received and transmitted.

closed 2021-04-12 by github-actions[bot]