Ysurac/openmptcprouter #1513: Download Hash

fuckFH, 2021-01-07:
## Expected Behavior
sha256 hash has to be identical with hash value which gets shown when I uploaded the file to the device.
I did it for: ext4 image: openmptcprouter-v0.57.3-r0+15225-bfc433efd4-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz (2021/01/05 - 70.81Mo - sha256sum)

## Current Behavior
It isn't identical, but the upgrade still works.

## Possible Solution
In general, it is often so that the developer forgets to update the hash on the download webpage.
A script on the webserver which calculates the hash from the published files automatically can be a solution.

##Coments
The project is great!

Ysurac, 2021-01-07:
sha256 hash are generated via a script and after download and check they are identical.

fuckFH, 2021-01-08:
Ok, I did a manual check of the SHA256 value and compared it with the value of the webpage and there was a difference. I have no idea why.
But when there is a mechanism on the running box which makes a comparison to the distribution webpage it is perfect.

closed 2021-01-08 by fuckFH