Ysurac/openmptcprouter #1517: Configuring with a USB 4G LTE modem no connection

kevmo314, 2021-01-10:
## Expected Behavior

ping or wget should work. I get the impression that this is a rather silly misconfiguration, but I've been trying to figure it out for a while now and not sure what's wrong... :( would be really grateful for the help

## Current Behavior

`ping google.com` doesn't result in any output. However, I think my LTE modem has ICMP disabled. I also tried wget'ing an ip however it can't connect.

Network status:

[image]

MPTCP interfaces:

[image]

USB interface:

[image]

`ip r` output:

```
root@OpenMPTCProuter:~# ip r
default via 192.168.10.1 dev wan1
default via 192.168.10.1 dev wan1 metric 3
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.10.0/24 dev wan1 scope link metric 3
192.168.10.0/24 dev usb0 proto static scope link metric 8
```

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: 5.4.81-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-01-10:
The interface is not working at all.
What is the dongle model ? Some create an ethernet interface (Huawei Hi-Link for example), some need to use modem manager protocol.

kevmo314, 2021-01-10:
I have a ZTE MF833V. The dongle itself works on a windows machine (I'm able to connect, sim works and everything). I noticed it registers as a cdc_ether interface on the rpi. Is there more config needed/additional drivers I need to download?

Ysurac, 2021-01-10:
For this dongle, as it's seems to be in right mode, you only need to set the interface as DHCP client. This should work.

kevmo314, 2021-01-10:
I see, I switched it over to DHCP client:

[image]

and updated the gateway for the WAN:

[image]

However the status page/symptoms still are unchanged: 

[image]

I also tried binding the WAN to 192.168.0.112 with no success.

Also, if the interface is on DHCP, wouldn't the WAN binding fail if the DHCP lease expires and the modem gets assigned to a new IP? Is there a way around that?

Ysurac, 2021-01-10:
Because you modified wan1 interface, and by default it's a MacVLAN interface. You need to switch it to normal in "Physical settings" tab in Network->Interface

kevmo314, 2021-01-11:
Hmm, so I did that, wan1 still doesn't seem to want to connect?

[image]

Ysurac, 2021-01-11:
The physical interface can't be wan1, set it to usb0

kevmo314, 2021-01-13:
Thanks for the help, once everything is configured I must say OpenMPTCProuter works exceptionally well! I'm using this for a small side business, hopefully I'll be in a position to contribute to your patreon soon :)

closed 2021-01-13 by kevmo314