Ysurac/openmptcprouter #1519: usb-modeswitch on v 57.4

claw2004, 2021-01-12:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

claw2004, 2021-01-12:
Hello. I had packet usb_modeswitch installed on v0.54 and had no problem with my usb stick modems. Few days ago I updated my router to v0.57 and usb-modeswitch packet does not installing on this version of firmware. Can You help me with advice for this situation?

Ysurac, 2021-01-12:
usb-modeswitch package is installed on v0.57 by default.

claw2004, 2021-01-12:
Then why don't my eth2 eth3 usb modems appear in the add wan interfaces tab, as it was after installing the usb-modeswitch on v0.54 package?

Ysurac, 2021-01-12:
Check logs in State menu.

claw2004, 2021-01-13:
Hello!
As I think the router does not switch the usb modem to zerocd mode, which the usb-modeswitch package did. 

Logs:

Telnet Log:

login as: root
BusyBox v1.31.1 () built-in shell (ash)
   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r0+15225-bfc433efd4)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.57.3

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Jan  5 07:01:48 UTC 2021
-----------------------------------------------------
root@OpenMPTCProuter:~# usb-modeswitch
-ash: usb-modeswitch: not found

That means what the usb-modeswitch does not installed on system!

I made in terminal:

root@OpenMPTCProuter:~# opkg update
root@OpenMPTCProuter:~# opkg install usb-modeswitch

Unknown package 'usb-modeswitch'.
Collected errors:
 * opkg_install_cmd: Cannot install package usb-modeswitch.

Syslog:

Wed Jan 13 01:09:29 2021 kern.info kernel: [  261.051765] usb 2-1: new high-speed USB device number 5 using xhci-hcd
Wed Jan 13 01:09:29 2021 kern.debug kernel: [  261.232625] usb 2-1: no of_node; not parsing pinctrl DT
Wed Jan 13 01:09:29 2021 kern.debug kernel: [  261.233443] usb-storage 2-1:1.0: no of_node; not parsing pinctrl DT
Wed Jan 13 01:09:29 2021 kern.info kernel: [  261.233468] usb-storage 2-1:1.0: USB Mass Storage device detected
Wed Jan 13 01:09:29 2021 kern.info kernel: [  261.242949] scsi host2: usb-storage 2-1:1.0
Wed Jan 13 01:09:30 2021 kern.notice kernel: [  262.262564] scsi 2:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
Wed Jan 13 01:09:30 2021 kern.debug kernel: [  262.270872] sd 2:0:0:0: no of_node; not parsing pinctrl DT
Wed Jan 13 01:09:30 2021 kern.notice kernel: [  262.271263] scsi 2:0:0:1: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Wed Jan 13 01:09:30 2021 kern.debug kernel: [  262.279716] sd 2:0:0:1: no of_node; not parsing pinctrl DT
Wed Jan 13 01:09:30 2021 kern.warn kernel: [  262.287082] sd 2:0:0:1: Power-on or device reset occurred
Wed Jan 13 01:09:30 2021 kern.notice kernel: [  262.293400] sd 2:0:0:1: [sda] Attached SCSI removable disk

Do you have any suggestions?

Ysurac, 2021-01-13:
As I said, usb-modeswitch package is installed. This package include usbmode tool, with json-based configuration, instead of classic usb_modeswitch.
What is the model of the key ?

claw2004, 2021-01-13:
Huawei E3372H-173

AT^FHVER
^FHVER:"E3372H-153 22.328.62.00.143,CL2E3372HM Ver.A"
OK

AT^VERSION?
^VERSION:BDT:Nov 28 2017, 08:53:20
^VERSION:EXTS:22.328.62.00.143
^VERSION:INTS:22.328.62.00.143
^VERSION:EXTD:WEBUI_17.100.15.02.25_RE5
^VERSION:INTD:WEBUI_17.100.15.02.25_RE5
^VERSION:EXTH:CL2E3372HM Ver.A
^VERSION:INTH:CL2E3372HM Ver.A
^VERSION:EXTU:E3372
^VERSION:INTU:E3372H-153
^VERSION:CFG:1004
^VERSION:PRL:
^VERSION:OEM:
^VERSION:INI:
OK

claw2004, 2021-01-13:
The second USB modem that does not switching to working state is Huawei E8372h 

AT^FHVER

^FHVER:"E8372H-153 21.328.03.00.00,CL1E8372HM Ver.A"
OK

AT^VERSION?

^VERSION:BDT:Oct 24 2017, 17:57:48
^VERSION:EXTS:21.328.03.00.00
^VERSION:INTS:21.328.03.00.00
^VERSION:EXTD:WEBUI_17.100.18.03.143_HILINK_Mod1.21
^VERSION:INTD:WEBUI_17.100.18.03.143_HILINK_Mod1.21
^VERSION:EXTH:CL1E8372HM Ver.A
^VERSION:INTH:CL1E8372HM Ver.A
^VERSION:EXTU:E8372
^VERSION:INTU:E8372H-153
^VERSION:CFG:1005
^VERSION:PRL:
^VERSION:OEM:
^VERSION:INI:E8372hCUST-B01C00
OK

Sys log:

Wed Jan 13 07:10:10 2021 kern.info kernel: [21909.584075] usb 1-1: new high-speed USB device number 9 using orion-ehci
Wed Jan 13 07:10:10 2021 kern.debug kernel: [21909.785531] usb 1-1: no of_node; not parsing pinctrl DT
Wed Jan 13 07:10:10 2021 kern.debug kernel: [21909.794581] usb-storage 1-1:1.0: no of_node; not parsing pinctrl DT
Wed Jan 13 07:10:10 2021 kern.info kernel: [21909.794609] usb-storage 1-1:1.0: USB Mass Storage device detected
Wed Jan 13 07:10:10 2021 kern.info kernel: [21909.802149] scsi host2: usb-storage 1-1:1.0
Wed Jan 13 07:10:11 2021 kern.notice kernel: [21910.835172] scsi 2:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
Wed Jan 13 07:10:11 2021 kern.debug kernel: [21910.843517] sd 2:0:0:0: no of_node; not parsing pinctrl DT
Wed Jan 13 07:10:11 2021 kern.notice kernel: [21910.844165] scsi 2:0:0:1: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Wed Jan 13 07:10:11 2021 kern.debug kernel: [21910.852458] sd 2:0:0:1: no of_node; not parsing pinctrl DT
Wed Jan 13 07:10:11 2021 kern.warn kernel: [21910.853266] sd 2:0:0:1: Power-on or device reset occurred
Wed Jan 13 07:10:11 2021 kern.notice kernel: [21910.860569] sd 2:0:0:1: [sda] Attached SCSI removable disk

Ysurac, 2021-01-13:
I have an Huawei E8372h and I don't have this problem.
I don't see idVendor and idProduct in your log lines, do you have them somewhere in the log ?

claw2004, 2021-01-13:
E3372 VID 12d1 PID 14dc
E8372 VID 12d1 PID 14db

claw2004, 2021-01-13:
Hello!
I updated the firmware of the modems with modified firmware where they are forced to switch to the desired mode of operation at startup. The eth1 and eth2 interfaces are appeared, but if you look at the logs, communication with modems only worked on the RNDIS protocol. I tried to force the CDC protocol for Linux but then the interfaces don't appear.

Syslog:

Eth2 modem connected:
Wed Jan 13 11:52:08 2021 kern.info kernel: [ 1913.962100] usb 2-1.2: new high-speed USB device number 17 using xhci-hcd
Wed Jan 13 11:52:08 2021 kern.debug kernel: [ 1914.092953] usb 2-1.2: no of_node; not parsing pinctrl DT
Wed Jan 13 11:52:08 2021 kern.debug kernel: [ 1914.095086] rndis_host 2-1.2:1.0: no of_node; not parsing pinctrl DT
Wed Jan 13 11:52:08 2021 kern.info kernel: [ 1914.096166] rndis_host 2-1.2:1.0 eth2: register 'rndis_host' at usb-f10f8000.usb3-1.2, RNDIS device, 0c:5b
Wed Jan 13 11:52:08 2021 daemon.notice netifd: Interface 'mega' is enabled
Wed Jan 13 11:52:08 2021 daemon.notice netifd: Network device 'eth2' link is up
Wed Jan 13 11:52:08 2021 daemon.notice netifd: Interface 'mega' has link connectivity
Wed Jan 13 11:52:08 2021 daemon.notice netifd: Interface 'mega' is setting up now
Wed Jan 13 11:52:08 2021 kern.debug kernel: [ 1914.111597] usb-storage 2-1.2:1.2: no of_node; not parsing pinctrl DT
Wed Jan 13 11:52:08 2021 kern.info kernel: [ 1914.111626] usb-storage 2-1.2:1.2: USB Mass Storage device detected
Wed Jan 13 11:52:08 2021 kern.info kernel: [ 1914.119735] scsi host3: usb-storage 2-1.2:1.2
Wed Jan 13 11:52:08 2021 daemon.notice netifd: mega (29043): udhcpc: started, v1.31.1
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p1506d0102dc00dsc00dp00icFFisc02ip16in01 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: wwan0 - INTERFACE: eth2
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p14DCd0102dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: eth1 - INTERFACE: eth2
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: eth2 - INTERFACE: eth2
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p1506d0102dc00dsc00dp00icFFisc02ip16in01 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: wwan0 - INTERFACE: eth2
Wed Jan 13 11:52:08 2021 daemon.notice netifd: mega (29043): udhcpc: sending discover
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p14DCd0102dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: eth1 - INTERFACE: eth2
Wed Jan 13 11:52:08 2021 user.notice OMR-Rename: dir:  - modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - chk_modalias: usb:v12D1p14DBd0102dc00dsc00dp00icE0isc01ip03in00 - ifname: eth2 - INTERFACE: eth2
Wed Jan 13 11:52:09 2021 kern.notice kernel: [ 1915.174977] scsi 3:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Wed Jan 13 11:52:09 2021 kern.debug kernel: [ 1915.183453] sd 3:0:0:0: no of_node; not parsing pinctrl DT
Wed Jan 13 11:52:09 2021 kern.warn kernel: [ 1915.184936] sd 3:0:0:0: Power-on or device reset occurred
Wed Jan 13 11:52:09 2021 kern.notice kernel: [ 1915.194239] sd 3:0:0:0: [sda] Attached SCSI removable disk
Wed Jan 13 11:52:11 2021 daemon.notice netifd: mega (29043): udhcpc: sending discover
Wed Jan 13 11:52:14 2021 daemon.notice netifd: mega (29043): udhcpc: sending select for 192.168.9.100
Wed Jan 13 11:52:14 2021 daemon.notice netifd: mega (29043): udhcpc: lease of 192.168.9.100 obtained, lease time 86400
Wed Jan 13 11:52:14 2021 daemon.debug dnsmasq[19232]: listening on eth2(#32): 192.168.9.100 port 53
Wed Jan 13 11:52:14 2021 daemon.notice netifd: Interface 'mega' is now up
Wed Jan 13 11:52:14 2021 user.notice firewall: Reloading firewall due to ifup of mega (eth2)
Wed Jan 13 11:52:16 2021 user.notice omr-bypass: Starting OMR-ByPass...

Eth1 modem connected:
Wed Jan 13 11:56:45 2021 kern.info kernel: [ 2191.311263] usb 2-1.3: new high-speed USB device number 18 using xhci-hcd
Wed Jan 13 11:56:45 2021 kern.debug kernel: [ 2191.442095] usb 2-1.3: no of_node; not parsing pinctrl DT
Wed Jan 13 11:56:45 2021 kern.debug kernel: [ 2191.443261] usb-storage 2-1.3:1.0: no of_node; not parsing pinctrl DT
Wed Jan 13 11:56:45 2021 kern.info kernel: [ 2191.443297] usb-storage 2-1.3:1.0: USB Mass Storage device detected
Wed Jan 13 11:56:45 2021 kern.info kernel: [ 2191.450104] scsi host2: usb-storage 2-1.3:1.0
Wed Jan 13 11:56:46 2021 kern.notice kernel: [ 2192.531952] scsi 2:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
Wed Jan 13 11:56:46 2021 kern.debug kernel: [ 2192.540413] sd 2:0:0:0: no of_node; not parsing pinctrl DT
Wed Jan 13 11:56:46 2021 kern.notice kernel: [ 2192.540836] scsi 2:0:0:1: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Wed Jan 13 11:56:46 2021 kern.debug kernel: [ 2192.549310] sd 2:0:0:1: no of_node; not parsing pinctrl DT
Wed Jan 13 11:56:46 2021 kern.warn kernel: [ 2192.549726] sd 2:0:0:1: Power-on or device reset occurred
Wed Jan 13 11:56:46 2021 kern.notice kernel: [ 2192.556580] sd 2:0:0:1: [sdb] Attached SCSI removable disk
Wed Jan 13 11:56:48 2021 kern.info kernel: [ 2194.092554] usb 2-1.3: USB disconnect, device number 18
[... cut ...]
Wed Jan 13 11:57:01 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jan 13 11:57:01 2021 daemon.info dnsmasq[19232]: read /etc/hosts - 4 addresses
Wed Jan 13 11:57:01 2021 daemon.info dnsmasq[19232]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jan 13 11:57:01 2021 daemon.info dnsmasq-dhcp[19232]: read /etc/ethers - 0 addresses
Wed Jan 13 11:57:01 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jan 13 11:57:01 2021 user.notice mptcp: Reloading mptcp config due to ifup of tele (eth1)

claw2004, 2021-01-15:
Rolled back to firmware version 0.54. Multipath does not work correctly

claw2004, 2021-04-16:
Thanks **xemles**

closed 2021-04-16 by claw2004