Ysurac/openmptcprouter #1520: Question: Load Balancing with MPTCP BLEST Scheduler possible?

ByteArrayException, 2021-01-12:
Good evening,

Is it possible to split the traffic on all routes with the BLEST scheduler? Or does this type of scheduler send all traffic to the best LowRTT route and only if this route cannot carry the traffic it will used an another uplink (concerns LTE)?

Best regards

Ysurac, 2021-01-15:
Not it's not possible. You can find the article that explain how BLEST work here: http://dl.ifip.org/db/conf/networking/networking2016/1570234725.pdf

ByteArrayException, 2021-01-15:
Hey,

Thanks for the feedback. Is there an alternative for load balancing with LTE modems? Or is BLEST the best option for dealing with LTE in terms of MPTCP?

Best regards

Ysurac, 2021-01-15:
In most case BLEST is the best scheduler, but you can try ECF or default. But none are doing load balancing, only roundrobin can do some load balancing but give bad performance.

ByteArrayException, 2021-01-15:
Hey,

Thank you very much, I will try these variants and get back to you here.

Best regards

ByteArrayException, 2021-01-16:
Hey,

I'm back. In fact, in my test scenarios BLEST was the best method, thank you very much already. With 4G/LTE connections, should you rather use cubic or bbr?

best regards

Ysurac, 2021-01-16:
Cubic is better when connections are not stable (it's the case with LTE), else BBR is better.
BBRv2 seems to give a quite good result in all case, but I don't have enough reports yet.

ByteArrayException, 2021-01-16:
Okay, I tested everything with cubic. Then I'll stick with it. Thank you very much!

closed 2021-01-16 by ByteArrayException