Ysurac/openmptcprouter #1521: Frequent VPN Tunnel Down / Disconnects

goggles24, 2021-01-12:
## Expected Behavior
VPN Tunnel Should Stay Up

## Current Behavior
I am experiencing fairly frequent VPN disconnection, observed on the MPTCP Status page.
I am currently working in remote production and relying heavily on accessing AWS instances via Teradici, when the VPN tunnel is down this connection drops for several seconds & response is very sluggish. 
The router seems perfectly reliable for my general browsing & downloading, but when I'm operating with higher loads it seems to disconnect maybe once an hour. I haven't been able to track down a specific activity/condition that triggers it. I will typically be connected to 1x AWS instance over teradici and on a couple of VC calls when this occours. Any help would be greatly appreciated.

VPS is running in a HostWorld Debian server
Local Router is a RPI4. 
I've been experiencing this for a few versions and behaviour has not changed.
Running 1x ADSL connection (master) 20mbps Down/ 3up
Plus 2 4g connections which are generally stable but somewhat variable as is to be expected.

Here are some logs from 2 disconnects I've been able to catch:
Thu Dec 17 16:49:59 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Dec 17 16:49:59 2020 daemon.info dnsmasq[7013]: read /etc/hosts - 4 addresses
Thu Dec 17 16:49:59 2020 daemon.info dnsmasq[7013]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Dec 17 16:49:59 2020 daemon.info dnsmasq-dhcp[7013]: read /etc/ethers - 0 addresses
Thu Dec 17 16:49:59 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Dec 17 16:49:59 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Dec 17 16:49:59 2020 daemon.err glorytun[16009]: 107.181.136.11.65001: shutdown
Thu Dec 17 16:50:00 2020 daemon.err /usr/bin/ss-redir[23539]: remote recv: Operation timed out
Thu Dec 17 16:50:00 2020 daemon.err glorytun[16009]: read: Operation timed out
Thu Dec 17 16:50:00 2020 daemon.info glorytun[16009]: STOPPED tun0
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Network device 'tun0' link is down
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Dec 17 16:50:00 2020 daemon.debug dnsmasq[7013]: stopped listening on tun0(#81): 10.255.255.2 port 53
Thu Dec 17 16:50:00 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Thu Dec 17 16:50:00 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Thu Dec 17 16:50:00 2020 daemon.info glorytun[2959]: INITIALIZED tun0
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec 17 16:50:00 2020 daemon.warn dnsmasq[7013]: failed to create listening socket for fe80::151b:a96d:947c:ad7f%tun0: Address not available
Thu Dec 17 16:50:00 2020 daemon.warn dnsmasq[7013]: failed to create listening socket for fe80::151b:a96d:947c:ad7f%tun0: Address not available
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Network device 'tun0' link is up
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Dec 17 16:50:00 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Thu Dec 17 16:50:00 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Thu Dec 17 16:50:00 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Thu Dec 17 16:50:00 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Dec 17 16:50:00 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu Dec 17 16:50:00 2020 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 172.31.128.129 dev br-test weight 1 nexthop via 192.168.8.1 dev br-threebridge weight 1 nexthop via 192.168.10.1 dev br-vodbridge weight 2
Thu Dec 17 16:50:02 2020 daemon.err /usr/bin/ss-redir[23541]: remote recv: Operation timed out
Thu Dec 17 16:50:03 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Dec 17 16:50:03 2020 daemon.err glorytun[2959]: 107.181.136.11.65001: connected
Thu Dec 17 16:50:03 2020 daemon.info glorytun[2959]: STARTED tun0

And another one:
Mon Jan  4 14:18:16 2021 daemon.info glorytun[26644]: STOPPED tun0
Mon Jan  4 14:18:16 2021 daemon.err glorytun[26644]: 107.181.136.11.65001: connected
Mon Jan  4 14:18:17 2021 user.notice post-tracking-post-tracking: wan2 (br-threebridge) switched off because ping from 192.168.8.117 error (80.67.169.40,114.114.114.114,1.1.1.1)
Mon Jan  4 14:18:17 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev br-threebridge
Mon Jan  4 14:18:25 2021 daemon.err glorytun[26644]: read: Connection reset by peer
Mon Jan  4 14:18:25 2021 daemon.err glorytun[26644]: 107.181.136.11.65001: key exchange failed
Mon Jan  4 14:18:25 2021 daemon.info glorytun[26644]: STOPPED tun0
Mon Jan  4 14:18:27 2021 user.notice post-tracking-post-tracking: wan2 (br-threebridge) switched up
Mon Jan  4 14:18:27 2021 daemon.err glorytun[26644]: 107.181.136.11.65001: connected
Mon Jan  4 14:18:28 2021 daemon.info glorytun[26644]: STARTED tun0

## Specifications

  - OpenMPTCProuter version:0.57.3 
  - OpenMPTCProuter VPS version: 0.1024 5.4.81-mptcp
  - OpenMPTCProuter platform: RPI4 4gb

Ysurac, 2021-01-13:
You seems to use some bridge interface, why ?

goggles24, 2021-01-13:
My 3 Wan interfaces come in on tagged vlans, and my lan interface is untagged, all on the RPI4 NIC.
When I initially set this up a few months a go the only way I seemed to be able to get connectivity was to create the Vlans as Bridge interface, and then set those bridge interfaces as the interface for my WAN connections. 
Could you confirm the correct setup for setting up my WAN connections?

Ysurac, 2021-01-13:
Using eth0.lanid as physicial interface should work without the need of any bridge.

goggles24, 2021-01-13:
Ok, so I have no idea why I couldn't set this up initially, I must have been doing something wrong! But I have just removed all the bridge interfaces and it is working straight away.
I guess I will monitor connectivity over the next few days to see if this stabilises the VPN tunnel.
Thank you!

goggles24, 2021-01-14:
Unfortunately I'm still seeing the disconnects, even having removed the bridge interfaces.
It seems to be linked to heavy upload traffic, if I upload a large video file to my company storage, it disconnects almost immediately.

Ysurac, 2021-01-15:
You can enable SQM and set download/upload speed of each connection, this should help.

goggles24, 2021-01-15:
I have enabled SQM for all connections (2 are LTE but thought I'd give it a go) and set the speeds appropriately, unfortunately it still disconnects almost instantaneously.
Is there a log I can pull to see what the cause of the disconnect is?
One thing I should mention is that to upload to this particular server I have to connect through my company VPN. However I also so see disconnects when not connected to that, so not sure it's related but it is a reliable way for me to repro the problem.

Ysurac, 2021-01-15:
State->System log will give you the reason.

goggles24, 2021-01-15:
There doesn't seem to be anything relevant shown in the log for the time period, in this example Disconnect was at 14:06 and reconnect at 14:11. Logging level is set at debug.

Ysurac, 2021-01-15:
This is a computer that ask an address to a DHCP server, the server offer an address and the computer doesn't answer and ask again. Not related if the mac is not the client computer.
How do you check disconnect ?

goggles24, 2021-01-15:
[image]
I check in System > OpenMPTCP Router> Status.

Ysurac, 2021-01-15:
And you have a problem ? This check is done using a ping with a small timeout.

goggles24, 2021-01-15:
Yes when this occours I experience significant disruption. Zoom etc will become unstable/unusable and critically for me my control of remote AWS instances becomes unstable. All of which I guess could be explained by a high latency which would break that check as well?

Ysurac, 2021-01-15:
Yes. It's why there is multiple proxy/vpn that can be used in the wizard. Some can give lower latency in some cases.
Some TCP Congestion Control in Network->MPTCP can also help.

goggles24, 2021-01-15:
Thank you for your support, I definitely have some reading up to do here! Do you have any recommendation of settings for VPN or Congestion Control that may help reduce latency?
Thanks again.

goggles24, 2021-01-15:
I have switched to OpenVPN, and fingers crossed this seemse to have solved the latency issues for now. 
Thanks again.

closed 2021-01-15 by goggles24

georgedorn, 2021-01-19:
Is there a guide for switching to OpenVPN somewhere?

goggles24, 2021-01-19:
I changed it via the wizard, by checking the Advanced Settings box, it will show you the VPN settings. From there you can change the default VPN to OpenVPN. I didn't have to do anything else for this to work.