Ysurac/openmptcprouter #1524: https-dns-proxy error: attempt to index local 'value' (a nil value)

gartz, 2021-01-15:
## Expected Behavior
Enable the Https-dns-proxy

## Current Behavior
Shows the error in browser:
```
/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:43: attempt to index local 'value' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:43: in function 'get_provider_name'
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:86: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1353: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>
```

## Possible Solution
Seems like that the provider name might return null instead of a valid value.

## Steps to Reproduce the Problem

  1. Go to the https-dns-proxy tab
  2. Click to Enable the proxy
  3. Click to start the service

## Context (Environment)
After executing the steps the error showed, doesn't matter to refresh or click again it always shows the error on that page.
Reboot the device didn't fix the issue.

## Specifications

  - OpenMPTCProuter version:  v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter platform: UEFI x86 ext4 version in a proxmox VM

Ysurac, 2021-01-15:
I will fix the error, but not sure that this will really fix all.
What do you have as result of command ```uci show https-dns-proxy``` from the router via SSH or via Services->Terminal ?

closed 2021-01-16 by Ysurac

gartz, 2021-02-02:
```
root@GartzRouter:~# uci show https-dns-proxy
https-dns-proxy.config=main
https-dns-proxy.config.update_dnsmasq_config='*'
https-dns-proxy.@https-dns-proxy[0]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[0].bootstrap_dns='8.8.8.8,8.8.4.4'
https-dns-proxy.@https-dns-proxy[0].user='nobody'
https-dns-proxy.@https-dns-proxy[0].group='nogroup'
https-dns-proxy.@https-dns-proxy[1]=https-dns-proxy
https-dns-proxy.@https-dns-proxy[1].bootstrap_dns='1.1.1.1,1.0.0.1'
https-dns-proxy.@https-dns-proxy[1].user='nobody'
https-dns-proxy.@https-dns-proxy[1].group='nogroup'
```

Ysurac, 2021-02-02:
The fix is working, no more errors if there is no listen address and port.
This will be available in next release.
Thanks.