Ysurac/openmptcprouter #1525: Multiple Huawei E3372H-320

thothbot, 2021-01-17:
## Expected Behavior
I have 4 modems Huawei E3372H-320. I need to change modems IP on router side instead of using modem's default in subnet 192.168.8.0/24

## Current Behavior
All of those modems have LAN subnet 192.168.8.0/24 which can't be changed. Modem can't be unlocked or new firmware can't be installed. WAN IP can be obtained over DHCP and it is the same for all WAN interfaces 192.168.8.100, gw 192.168.8.1

## Specifications

  - OpenMPTCProuter version: v0.57.3
  - OpenMPTCProuter VPS version: v0.1024
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-01-17:
And what is the issue ? It's not possible to change modem IP there is no way to change them on router side. The only think that can be modified is to set WAN interface as static and set an IP in 192.168.8.x range and use 192.168.8.1 as gw each times.

thothbot, 2021-01-17:
That's not an issue when we use one modem per router. We can get IP address on WAN interface by DHCP or static.

I want to use 4 modems simultaneously.
If I connect all 4 modems I'll get the following nets:

```
wan1: eth1 -> 192.168.8.0/24
wan2: eth2 -> 192.168.8.0/24
wan3: eth3 -> 192.168.8.0/24
wan4: eth4 -> 192.168.8.0/24
```

which is not working configuration

I need to add 4 nets and route all traffic from 192.168.8.0/24 to new virtual net and make wan interface based on that net.

So the question is how can I do this? Can I use pseudo bridge or something?

closed 2021-04-22 by github-actions[bot]