Ysurac/openmptcprouter #1527: Single WAN setup (OMR to bypass CG-NAT)

omgbathtub, 2021-01-18:
## Expected Behavior
Wanting a low maintenance no frills setup. My ISP is a hotspot from an LTE provider. I'm looking to use OMR as a way to bypass the provider's CGNAT and my public IP VPS as my exit/entry point for my network. I technically have no need for aggregating connections, but OMR makes port forwarding a breeze so I use it for that reason.

## Current Behavior
I have my hotspot USB tethered to my Pi 4B w/ OMR and it is the one and only WAN I'll use. I'm curious what proxy/VPN protocol will have the least overhead/least latency and also supports all VPS ports being forwarded to OMR. Glorytun UDP seems to give me the least latency but seemingly at random, the status page will show the tunnel as being down, and I'll lose my port forwards as it fallsback to just the proxy with no VPN.

## Specifications

  - OpenMPTCProuter version: 0.57.3
  - OpenMPTCProuter VPS version: 0.1014
  - OpenMPTCProuter platform: RPi4B

TalalMash, 2021-01-19:
OMR is pretty buggy as is still in alpha state, I would recommend to reset, change congestion control to lia for low bufferbloat. Reset every time you change the same setting back and forth few times.
Glorytun UDP with 0.57.3 is currently broken, you have to move "fixed" and "auto" line in post-tracking file after tx and rx.
I also recommend rebooting the VPS before rebooting your OMR client on every change you make, sometimes MPTCP settings don't apply on the server side when setting from OMR.
OMR seems to be overkill, you can simply use Wireguard with PostUp forwarding all ports, and use OpenWRT's firewall interface for port forwarding on the client.

omgbathtub, 2021-01-19:
I appreciate the reply! With OMR, I have a single NAT to the interface, with wireguard I was under the impression it would be double NAT since it is layer 3 unlike the tunnels like OMR use are layer 2. Am I incorrect? I have not the slightest clue of what all PostUp and PostDown rules I would need, I'm a novice with linux networking

TalalMash, 2021-01-19:
AFAIK it's single NAT for TCP only. I won't clutter this issue more regarding Wireguard.

closed 2021-04-25 by github-actions[bot]