Ysurac/openmptcprouter #1528: Wifi Invité directement sur OMR?

JeedHome44, 2021-01-18:
Bonjour,

Depuis que j'ai passé mes 2 WAN en MacVLAN, tout marche à merveille!
Encore merci de m'avoir aidé quand ça n'allait pas du tout!

Je viens aujourd'hui savoir s'il est possible de gérer un wifi invité directement dans le logiciel OMR sur le Raspberry? J'ai un wifi invité qui ne marche plus sur l'un de mes points d'accès. Je me demandais donc s'il n'était pas plus simple de faire un wifi directement sur OMR et le configurer pour que tous les appareils connectés directement en wifi sur OMR ne voient pas mon réseau LAN et aient accès uniquement à internet.

J'ai bien conscience que le wifi du Raspberry n'est pas top mais c'est juste pour les invités qui viennent de temps en temps à la maison et leur éviter d'utiliser leur 4G.

Est-ce possible? Si oui, pouvez vous me dire comment le configurer?

Merci beaucoup

JeedHome44, 2021-01-18:
Je pense avoir fait une boulette en essayant de faire un wifi invité.
Sous le nom du wifi, ça proposait les différentes interfaces OMR.
J'ai pensé qu'en mettant omrvpn ca voulait dire que je voulais que le wifi envoie les appareils que sur le VPN mais ça m'a mis un bordel.
J'ai vu le mot bridge et j'ai quand même validé.
Donc la, j'ai l'interface omrvpn qui est totalement déprogrammé j'ai l'impression.
Je l'ai décoché du sans fil et j'ai désactivé le wifi sans fil mais ça n'a pas l'air d'avoir rétablit les paramètres d'avant.
Possible de me dire comment était configuré de base l'interface omrvpn et celle dessous?
Merci

JeedHome44, 2021-01-18:
Tout est au vert sur OMR sauf en rouge le routeur qui affiche "Le VPN n'est pas lancé et l'interface omrvpn qui dit que le dispositif n'est pas connecté.
En attribuant cette interface au wifi ca a du mettre le bordel quelque part? Mais ou...

J'espère que vous pourrez me dire...

Je suis vraiment pas doué avec OMR.

Je coche toujours des trucs qu'il ne faut pas.

Ysurac, 2021-01-19:
Il y a cette doc sur OpenWRT: https://openwrt.org/docs/guide-user/network/wifi/guestwifi/guest-wlan
Tout peut se faire via l'interface web, ce qui est important c'est de configurer l'interface wifi sur un autre réseau que ceux présent et de définir une nouvelle zone dans le firewall.

JeedHome44, 2021-01-19:
Ça m'a l'air un peu complexe tout ça.
J'ai nommé le wifi de OMR "Invités" et j'ai activé le mot de passe. J'ai coché "cacher les clients".
Mais côté Pare-feu je ne vois pas quoi faire et le wifi n'apparaît pas sur mon téléphone même activé sur OMR

Ysurac, 2021-01-19:
Commence par suivre la doc: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

kevin39, 2021-01-21:
L'idéal est à mon sens est quand même d'utiliser OMR juste pour du routing et de laisser le wifi à un appareil dédié qui fera cela beaucoup mieux. Le rpi n'est pas adapté pour avoir un wifi de qualité, comparé à un bon routeur wifi et ses antennes ^^

Donc moi je te conseillerai plutôt d'investir dans un bon routeur wifi que tu connectes sur ton switch.
Tu désactives le DHCP sur ce routeur surtout et c'est réglé.

JeedHome44, 2021-01-21:
Bah c'est ce qui est fait la mais mon wifi Guest ne permet plus d'avoir internet. Je pense que mon point d'accès a du le bloquer et j'ai plus accès à ce point d'accès sans raison. C'est un TP link Archer C9 et si je le paramètre avec un pc, une fois en point d'accès il est invisible sur le réseau.
J'essaie d'y accéder avec 192.168.1.1 mais ça ne répond pas.

closed 2021-01-30 by JeedHome44

JeedHome44, 2021-05-20:
Bonjour,
Je réactive l'issue car j'ai pris des routeurs Xiaomi. Impossible de les mettre en mode routeur et de désactiver leur DHCP. Je dois obligatoirement les mettre en mode répéteur filaire.
Mais dans ce mode, le wifi invité est désactivé.

Je pense donc me repencher sur comment faire un wifi invité sur OMR. Ça serait juste pour de la famille qui vient de temps en temps donc pas folichon mais c'est le but recherché.

Je vois dans l'interface OMR du wifi qu'il y a une option pour isoler chaque client. Ça permet qu'ils ne voient pas le LAN et donc faire comme un wifi invité?
C'est bien ça?