Ysurac/openmptcprouter #1529: Kernel Not able to boot on kvmvps@serversgalore.com.au

metro2450, 2021-01-19:
## Expected Behavior
finish installation on vps, reboot into system, can provide service to routers

## Current Behavior
Boot stuck at "Finished update UTMP about system Boot/Shutdown. "

## Specifications
[image]
 KVM 1G RAM tried virtio driver and ide/intel pro 1000 all fail at boot

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2021-01-22:
I think a module can't be loaded, but I can't really help here.
I would need to rent a VPS, they accept Paypal ?

metro2450, 2021-01-22:
> I think a module can't be loaded, but I can't really help here.
> I would need to rent a VPS, they accept Paypal ?

Recommended install is on Debian as indicated in the documentation. Ubuntu install is working in most case but some strange install may cause issues.

_Originally posted by @Ysurac in https://github.com/Ysurac/openmptcprouter/issues/1504#issuecomment-755232581_

After read this I have reinstalled Debian 10 x64 and everything is working perfectly.  Thank you very much :-)

Ysurac, 2021-01-23:
ok cool :)
I can add serversgalore.com.au as supported VPS so ? I don't have many in australia

closed 2021-04-29 by github-actions[bot]