Ysurac/openmptcprouter #1530: DHCP client on macvlan interfaces

anp369, 2021-01-19:
Is it possible to create macvlan interfaces with dhcp client enabled?
I want to achieve something simliar to #858.

## Expected Behavior

When creating interfaces I can select macvlan and use the dhcp client on it, so that I don't have to specify an static IP (which I don't always know) 

## Current Behavior

instead I only can assign a static IP via the wizard. If I add an macvlan interface via the 'Interfaces' section the interface doesn't get an valid IP. In the router's logs that should assign the IP I can see DHCPDISCOVER messages from the macvlan interface but it doesn't seem to respond to it (no DHCPACK).

```
Tue Jan 19 14:13:38 2021 daemon.info dnsmasq-dhcp[4838]: DHCPDISCOVER(br-lan) 12:eb:f1:a9:be:da
Tue Jan 19 14:13:38 2021 daemon.info dnsmasq-dhcp[4838]: DHCPOFFER(br-lan) 192.168.1.117 12:eb:f1:a9:be:da
Tue Jan 19 14:13:42 2021 daemon.info dnsmasq-dhcp[4838]: DHCPDISCOVER(br-lan) 12:eb:f1:a9:be:da
Tue Jan 19 14:13:42 2021 daemon.info dnsmasq-dhcp[4838]: DHCPOFFER(br-lan) 192.168.1.117 12:eb:f1:a9:be:da
Tue Jan 19 14:13:44 2021 daemon.info dnsmasq-dhcp[4838]: DHCPDISCOVER(br-lan) 12:eb:f1:a9:be:da
Tue Jan 19 14:13:44 2021 daemon.info dnsmasq-dhcp[4838]: DHCPOFFER(br-lan) 192.168.1.117 12:eb:f1:a9:be:da
```
## Specifications

  - OpenMPTCProuter version: 0.57.3-r0
  - OpenMPTCProuter VPS version: v0.1024
  - OpenMPTCProuter platform: x86_64 in VirtualBox

Ysurac, 2021-01-19:
You can't, and this would be a strange network configuration.

anp369, 2021-01-19:
It is indeed a very strange setup. But on my OpenWRT router it works with the kmod-macvlan package. But I don't know how this is handled internally in openwrt. my provider is capping the speed per mac address and with openwrt I can open multiple interfaces with different Mac adresses and thus load-balance between them. Since this is not bonding I was wondering if this could be done with OMR to utilize all my interfaces.

Ysurac, 2021-01-19:
By default, interfaces are used based on latency. Should be possible to use Round Robin Multipath TCP scheduler.
As you are using virtualbox, you can create multiple interfaces in virtualbox and use them in OpenMPTCProuter.

anp369, 2021-01-19:
Virtualbox was only for testing if it would work in general since I have to custom compile it for my router. When I multiple dedicated interfaces in virtualbox add it using the normal mode into OMR it works perfectly.
The problem with this test setup was that the macvlan interfaces didn't accept the ip addresses they got from the dhcp server and that would be needed for my final usecase

TalalMash, 2021-01-19:
It's working here when configuring it from Networks page, not the Wizard. Pretty useful for on demand mobile hotspot connection (random ip).

anp369, 2021-01-20:
After some investigations I got DHCP working on macvlan interfaces. I think it was the fault of virtualbox' network adapters.
I know get an IP from dhcp and can connect to my vps. Thanks for your help, appreciated it :)

closed 2021-01-20 by anp369

TalalMash, 2021-01-20:
> After some investigations I got DHCP working on macvlan interfaces. I think it was the fault of virtualbox' network adapters.
> I know get an IP from dhcp and can connect to my vps. Thanks for your help, appreciated it :)

Note when using Virtualbox, MAC address spoofing only works with PCNETFast (100mbps)  and Intel Server MT (or T) (gigabit)