Ysurac/openmptcprouter #1531: TCP WAN switching performance using Master/Backup config?

luisglass, 2021-01-20:
Is there any experience with expected WAN switching performance for this implementation, when using a Master/Backup config?  

My use case is a GEO satellite connection for WAN2 and an LTE cellular connection for WAN1. Since aggregating such disparate capacity/latency links does not perform well. I configured cellular (WAN1) as Master and satellite (WAN2) as Backup. After starting a TCP session on WAN1, and failed the cellular link, the measured packet interruption time was about 2+ seconds, as the TCP session transitions from WAN1 to WAN2.  

Given the sub-flows are already be prepared on both links, I was thinking the packet switching time would be much faster (<1 second)

Thanks for any insights.

TalalMash, 2021-01-20:
My testing showed that the switching is immediate with redundant mode only, Glorytun UDP is the only instant with backup mode.

luisglass, 2021-01-25:
Thanks. Can you clarify how you tested? I used a flent voip-rrul test. The rrul switched with about a 2 second pause in throughput. The voip test did not transition

closed 2021-05-01 by github-actions[bot]