Ysurac/openmptcprouter #1532: Is dnsmasq vulnerability affecting omr?

Malaga82, 2021-01-21:
Hello,
i read about a dnsmasq vulnerability, below the links.

https://forum.openwrt.org/t/security-advisory-2021-01-19-1-dnsmasq-multiple-vulnerabilities/85903
https://kb.cert.org/vuls/id/434904

is this vulnerability affecting omr too?
Thanks

Adorfer, 2021-01-21:
The answer is: yes. 
(Case closed.)

Ysurac, 2021-01-21:
The answer is a little bit more complicated: by default, LAN devices ask resquest to dnsmasq but dnsmasq forward request to unbound. So unbound is the DNS client to internet. 
So DNS reply are validated by unbound before forwarded to dnsmasq, this should mitigate DNS cache poisoning vulnerabilities. 
DNSSEC is not enabled by default, so OpenMPTCProuter is not impacted by these vulnerabilities.

So, with default configuration, no vulnerability coming from Internet should affect omr. Only the CVE-2020-25686 can affect but this must be launched from the LAN.

closed 2021-04-26 by github-actions[bot]