Ysurac/openmptcprouter #1533: "Le VPN ne répond pas "

Bug-Storm, 2021-01-22:
Bonjour ,

J'ai un souci avec le VPN , ma config fonctionne nickel hors le VPN ,
[image]

```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].sequential_ip='1'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
[... 160 lines cut ...]
dscp.@domains[39].name='windowsupdate.com'
dscp.@domains[39].class='cs1'
dscp.@domains[39].comment='WindowsUpdate'
dscp.@domains[40]=domains
dscp.@domains[40].name='update.microsoft.com'
dscp.@domains[40].class='cs1'
dscp.@domains[40].comment='WindowsUpdate'
dsvpn.vpn=dsvpn
dsvpn.vpn.dev='tun0'
dsvpn.vpn.localip='10.255.251.2'
dsvpn.vpn.remoteip='10.255.251.1'
dsvpn.vpn.mode='client'
dsvpn.vpn.host='193.7xxxxxx'
dsvpn.vpn.key='6DB615206F6CF60F3D234BC2D0D6E933Axxxxxxxxxxxxxxxxxxxxxx'
dsvpn.vpn.port='65401'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=t…

```

Version du micrologiciel：openmptcprouter v0.57.3 r0+15225-bfc433efd4 RPI4B
Version 0.1024 5.4.86-mptcp VPS

jfh75, 2021-01-22:
Moi j'ai pareil dès que je réessaie d'utiliser V2ray. Si je remets shadowsocks + glorytun ça remarche. J'ai des dizaines de mail OMRVPN is down. Avec les mêmes version à jour que toi.

Ysurac, 2021-01-22:
@Bug-Storm, il n'y a pas toute la configuration.

Bug-Storm, 2021-01-22:
> @Bug-Storm, il n'y a pas toute la configuration.

`dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].sequential_ip='1'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001,65301'
dscp.@classify[4].comment='OMR vpn'
dscp.@domains[0]=domains
[... 1496 lines cut ...]
v2ray.main_reverse.bridges='omrbridge|omr.lan'
v2ray.omrexit=outbound
v2ray.omrexit.protocol='freedom'
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.country='FR'
wireless.radio0.cell_density='0'
wireless.radio0.hwmode='11g'
wireless.radio0.channel='auto'`

Ysurac, 2021-01-22:
l'interface omrvpn est définis en tant que "bridge". Ce n'est pas normal.

Bug-Storm, 2021-01-22:
> l'interface omrvpn est définis en tant que "bridge". Ce n'est pas normal.

Comment puis je modifier? Bizarre

Ysurac, 2021-01-22:
Dans Réseau->Interface puis éditer omrvpn et onglet paramètres physique mettre sur normal.

Bug-Storm, 2021-01-22:
> Dans Réseau->Interface puis éditer omrvpn et onglet paramètres physique mettre sur normal.

Nickelll merci!! par contre avec la vpn, Le MQTT et ssh peuvent fonctionner encore en "local (hors 4G) " ?

Ysurac, 2021-01-22:
Je n'ai pas compris ce que tu veux dire.

Bug-Storm, 2021-01-22:
> Je n'ai pas compris ce que tu veux dire.

en local(avec la livebox ) j'ai un rasp avec weewx (donc un MQTT +un homeassistant et bien sûr le ssh, tout  ça est redirectionner avec  No-Ip), est ce qu'avec le VPN activé , je vais pouvoir toujours faire la redirection des ports

Ysurac, 2021-01-22:
Actuellement la redirection depuis un WAN ne fonctionne pas, j'en cherche la raison.
La redirection depuis le VPS vers le LAN fonctionne.

Bug-Storm, 2021-01-22:
> Actuellement la redirection depuis un WAN ne fonctionne pas, j'en cherche la raison.
> La redirection depuis le VPS vers le LAN fonctionne.

Et juste la dernier chose xD Désolé, il est possible d'utiliser DyDDNS(No-Ip)? même la redirection depuis wan ne fonctionne pas :p

Ysurac, 2021-01-22:
Surement, à l'aide d'OMR-ByPass si c'est pour le mettre sur un wan ou en l'installant sur le routeur (doit y avoir moyen, ça fait longtemps que j'ai pas regardé DyDDNS), mais l'intérêt est faible.
Sinon possible en utilisant l'IP du VPS sans soucis.

Bug-Storm, 2021-01-23:
Coucou j'en profite vu qu'il est encore ouvert, j'ai un souci avec, par exemple si j'essaie de faire un speed test, la connexion avec le vps et vpn ne fonctionne plus, ça n'arrête pas d'avoir des deconnexion,

Bug-Storm, 2021-01-23:
`
Sun Jan 24 00:21:48 2021 daemon.err dsvpn[10808]: RTNETLINK answers: Permission denied
Sun Jan 24 00:21:48 2021 daemon.info dsvpn[10808]: Connected
Sun Jan 24 00:21:48 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jan 24 00:21:49 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 24 00:21:51 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jan 24 00:21:51 2021 daemon.info dnsmasq[5826]: read /etc/hosts - 4 addresses
Sun Jan 24 00:21:51 2021 daemon.info dnsmasq[5826]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Jan 24 00:21:51 2021 daemon.info dnsmasq-dhcp[5826]: read /etc/ethers - 0 addresses
Sun Jan 24 00:21:51 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 24 00:21:51 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jan 24 00:21:57 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.251.1 dev tun0
`

J'ai un permission Denied tout temps :/

Ysurac, 2021-01-24:
Car dsvpn est utilisé et je pense qu'il y a déjà un VPN sur la même interface. Les VPNs doivent être changés via Système->OpenMPTCProuter et pas le menu VPN pour éviter ça. Si c'est fait via le menu VPN faut pas oublier de virer le précédent.

Bug-Storm, 2021-01-24:
> Car dsvpn est utilisé et je pense qu'il y a déjà un VPN sur la même interface. Les VPNs doivent être changés via Système->OpenMPTCProuter et pas le menu VPN pour éviter ça. Si c'est fait via le menu VPN faut pas oublier de virer le précédent.

Ça fonctionne maintenant, donc c'était sûrement un bug, car hier j'avais réinstaller omr sur le vps, par contre,

Est ce qu'il existe un une option où si le ping dépasse une certaine valeur (ex: 200ms) la connexion s'arrête ? 🤔 Car en gros a chaque 5m j'ai un autre rasp avec weewx qui fait upload plusieurs fichiers et cela fait remonter le ping, et donc parfois ça dépasse les 200ms

Ysurac, 2021-01-24:
Tu peux vérifier dans Etat->Journal système si quelque chose est détecté. C'est OMR-Tracker qui s'occupe de ça, et tout est configurable dans Services->OMR-Tracker

Bug-Storm, 2021-01-24:
`
Sun Jan 24 10:04:32 2021 daemon.info dnsmasq-dhcp[5138]: read /etc/ethers - 0 addresses
Sun Jan 24 10:04:32 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 24 10:05:38 2021 user.notice OMR-VPS: Can't get Vps token, try later
Sun Jan 24 10:05:41 2021 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun Jan 24 10:05:41 2021 user.notice post-tracking-post-tracking: DSVPN down, restart it
Sun Jan 24 10:05:41 2021 daemon.info dsvpn[30694]: Done.
Sun Jan 24 10:05:41 2021 daemon.notice netifd: Network device 'tun0' link is down
Sun Jan 24 10:05:41 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Jan 24 10:05:41 2021 daemon.debug dnsmasq[5138]: stopped listening on tun0(#113): 10.255.251.2 port 53
Sun Jan 24 10:05:41 2021 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Jan 24 10:05:41 2021 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sun Jan 24 10:05:41 2021 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sun Jan 24 10:05:42 2021 daemon.err omr-tracker[1997]: Device "tun0" does not exist.
Sun Jan 24 10:05:42 2021 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Jan 24 10:05:44 2021 daemon.info dsvpn: starting dsvpn vpn instance vpn
Sun Jan 24 10:05:44 2021 daemon.info dsvpn[16752]: Interface: [tun0]
Sun Jan 24 10:05:44 2021 daemon.info dsvpn[16752]: Trying to reconnect
Sun Jan 24 10:05:44 2021 daemon.info dsvpn[16752]: Connecting to 193.70.xxxxx:65401...
Sun Jan 24 10:05:44 2021 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Jan 24 10:05:44 2021 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Jan 24 10:05:44 2021 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Jan 24 10:05:44 2021 daemon.notice netifd: Network device 'tun0' link is up
Sun Jan 24 10:05:44 2021 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Jan 24 10:05:44 2021 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sun Jan 24 10:05:44 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sun Jan 24 10:05:44 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sun Jan 24 10:05:44 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jan 24 10:05:45 2021 daemon.debug dnsmasq[5138]: listening on tun0(#114): 10.255.251.2 port 53
Sun Jan 24 10:05:45 2021 daemon.err dsvpn[16752]: RTNETLINK answers: Permission denied
Sun Jan 24 10:05:45 2021 daemon.info dsvpn[16752]: Connected
Sun Jan 24 10:05:51 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.251.1 dev tun0
Sun Jan 24 10:05:53 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 24 10:05:55 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jan 24 10:05:55 2021 daemon.info dnsmasq[5138]: read /etc/hosts - 4 addresses
Sun Jan 24 10:05:55 2021 daemon.info dnsmasq[5138]: read /tmp/hosts/odhcpd - 0 addresses
Sun Jan 24 10:05:55 2021 daemon.info dnsmasq[5138]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Sun Jan 24 10:05:55 2021 daemon.info dnsmasq-dhcp[5138]: read /etc/ethers - 0 addresses
Sun Jan 24 10:05:55 2021 user.notice omr-bypass: OMR-ByPass is running
`

Sur OMR-Tracker ->Paramètres par default  : je laisser le type par dns au lieu de ping , mais j'ai toujours des souci  et toujours le " RTNETLINK answers: Permission denied" alors que sur l'interface vpn , tous sont desactivé , sauf le dsvpn bien sûr

Ysurac, 2021-01-24:
Tu peux faire un truc du genre ```ps w | grep $(grep ^iff: /proc/*/fdinfo/* | cut -d\/ -f3)``` via SSH pour voir ce qui utilise l'interface tun0

Bug-Storm, 2021-01-24:
> ps w | grep $(grep ^iff: /proc/*/fdinfo/* | cut -d\/ -f3)

[image]

Ysurac, 2021-01-24:
Oui donc que par dsvpn, je dirais qu'il faut tenter un reboot.

Bug-Storm, 2021-01-24:
> Oui donc que par dsvpn, je dirais qu'il faut tenter un reboot.

Justement , je viens de re-flasher l'image sur le raspberry pi, pour le moment ça fonctionne nickel ;) ,avec la 4G Off  , la livebox prends le relais et avec la livebox off la 4G prends le relais, cote vps et vpn, pas de souci pour le moment !! je vais laisser le sujet encore ouvert jusqu'à demain ,au cas où j'ai un souci ;) ..

Bug-Storm, 2021-01-27:
@Ysurac  coucou, donc suis de retour, pour mon souci de port, c'était le firewall du vps, la je viens d'ouvrir le port et ça fonctionne, par contre je n'ai plus accès au vpn, à t-il un port spécifique pour le shackdown ou v2ray ? Merci

Ysurac, 2021-01-27:
Les ports sont listés ici : https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

Bug-Storm, 2021-01-27:
> Les ports sont listés ici : https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS

Merci :) , donc finalement c'est bien le OMR qui ne redirectionne pas les port correctement.. avec l'example sur [https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding](url)

le port 8000(source) - 80(dest) ça ne fonctionne pas , il faut mettre 80(source) - 80(dest) et c'est bizarre car en tcp c'est du random :/ ..

puis j'essai de faire de faire la regle NAT et pareil , ça ne fonctionne pas.

J'avais essayé avec mon rasp avec apache donc avec l'ip:80 j'ai `mis trop de temps à répondre`.. 

est il un bug ?

Bug-Storm, 2021-01-29:
@Ysurac, ça fonctionne nickel maintenant, fallait laisser l'option de redirection des  ports vers le vps activé puis créé la règle Nat ! Je viens de laisser mon raspberry avec weewx connecté toute la soirée, et pas de souci !! En tout cas merci pour ton projet 😉..

closed 2021-01-29 by Bug-Storm