Ysurac/openmptcprouter #1534: I always need to reconnect or restart modems...

maksaidd, 2021-01-22:
## Expected Behavior
Connecting 3 or 4 huawei modems and they're starting to work automatically

## Current Behavior
Every time when I start my Raspberry Pi4 with MPTCP Router, modems (2 or 3 of them) are not connected properly (no IP, Gateway down etc) and I need to restart them or unplug and plug again. 

## Specifications

  - OpenMPTCProuter version: 0.57.1
  - OpenMPTCProuter VPS version: 0.1023 5.4.81
  - OpenMPTCProuter platform: RP4 4Gb

Thank you for you great work! System is working! But, as I wrote, I have some problems with modem to MPTCP connection.
All settings are right. Static IP and no DHCP in modems (3272 and 3372 models). Right settings in MPTCP Wizard. But modems start to work after some "plugging-unplugging and restarting dances". I've found out that sometimes changing of ETH numbers helps. Is it a reason? How to fix right eth number (e.g eth1 or eth3) for particular modem?

maksaidd, 2021-01-22:
Screenshot after changing eth numbers [image]

Ysurac, 2021-01-22:
You can check State->System log and State->Kernel log.

maksaidd, 2021-01-22:
> ## Expected Behavior
> Connecting 3 or 4 huawei modems and they're starting to work automatically
>
> ## Current Behavior
> Every time when I start my Raspberry Pi4 with MPTCP Router, modems (2 or 3 of [...]

maksaidd, 2021-01-22:
.
Системный журнал
Sun Jan 17 09:59:25 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Jan 17 09:59:26 2021 user.notice OMR-VPS: Can't get vps token, try later
Sun Jan 17 09:59:27 2021 user.notice omr-bypass: Restart dnsmasq...
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[16666]: exiting on receipt of SIGTERM
Sun Jan 17 09:59:27 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 17 09:59:27 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 17 09:59:27 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: started, version 2.82 cachesize 150
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: DNS service limited to local subnets
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq-dhcp[32207]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain test
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain onion
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain localhost
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain local
Sun Jan 17 09:59:27 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain invalid
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain bind
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain use-application-dns.net
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using only locally-known addresses for domain lan
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using nameserver 127.0.0.1#5353
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: using 1 more local addresses
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: read /etc/hosts - 4 addresses
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq[32207]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jan 17 09:59:27 2021 daemon.info dnsmasq-dhcp[32207]: read /etc/ethers - 0 addresses
Sun Jan 17 09:59:27 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 17 09:59:29 2021 user.notice omr-bypass: Reload dnsmasq...
Sun Jan 17 09:59:29 2021 daemon.info dnsmasq[32207]: read /etc/hosts - 4 addresses
Sun Jan 17 09:59:29 2021 daemon.info dnsmasq[32207]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Sun Jan 17 09:59:29 2021 daemon.info dnsmasq-dhcp[32207]: read /etc/ethers - 0 addresses
Sun Jan 17 09:59:29 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 17 09:59:29 2021 daemon.debug dnsmasq[32207]: listening on tun0(#25): 10.255.255.2 port 53
Sun Jan 17 09:59:43 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: wan2 (eth2) switched off because gateway down and ping from 192.168.102.12 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.102.1 dev eth2
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: wan4 (eth4) switched off because gateway down and ping from 192.168.104.14 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: wan3 (eth3) switched off because gateway down and ping from 192.168.103.13 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: wan1 (eth1) switched off because gateway down and ping from 192.168.101.11 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.103.1 dev eth3
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.104.1 dev eth4
Sun Jan 17 09:59:46 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.101.1 dev eth1
Sun Jan 17 09:59:47 2021 user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps, no server API answer)
Sun Jan 17 09:59:48 2021 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Jan 17 09:59:48 2021 user.notice omr-tracker: Launching...
Sun Jan 17 09:59:53 2021 user.notice omr-tracker: Launched
Sun Jan 17 09:59:54 2021 user.notice omr-bypass: Starting OMR-ByPass...
Sun Jan 17 09:59:55 2021 user.notice omr-bypass: Restart dnsmasq...
Sun Jan 17 09:59:55 2021 daemon.info dnsmasq[32207]: exiting on receipt of SIGTERM
Sun Jan 17 09:59:55 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Jan 17 09:59:55 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Jan 17 09:59:55 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sun Jan 17 09:59:55 2021 user.notice omr-bypass: OMR-ByPass is running
Sun Jan 17 09:59:55 2021 daemon.info dnsmasq[4804]: started, version 2.82 cachesize 150
Sun Jan 17 09:59:55 2021 daemon.info dnsmasq[4804]: DNS service limited to local subnets
[... cut ...]
Fri Jan 22 15:36:51 2021 daemon.info dnsmasq[30412]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Fri Jan 22 15:36:51 2021 daemon.info dnsmasq-dhcp[30412]: read /etc/ethers - 0 addresses
Fri Jan 22 15:36:51 2021 user.notice omr-bypass: OMR-ByPass is running
Fri Jan 22 15:36:51 2021 user.notice mptcp: Reloading mptcp config due to ifup of wan4 (eth3)
Fri Jan 22 15:36:52 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Powered by openmptcprouter v0.57.1 r0+15225-bfc433efd4

maksaidd, 2021-01-22:
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83 (runner@fv-az128-843) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+15225-bfc433efd4)) #0 SMP Tue Dec 15 19:35:00 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:AF:04:3B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 3828792K/4050944K available (7038K kernel code, 430K rwdata, 1840K rodata, 512K init, 539K bss, 156616K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000157] Console: colour dummy device 80x25
[    0.000394] printk: console [tty1] enabled
[    0.000427] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000449] pid_max: default: 32768 minimum: 301
[    0.000642] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000726] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001373] Disabling memory control group subsystem
[    0.002059] ASID allocator initialised with 32768 entries
[    0.002149] rcu: Hierarchical SRCU implementation.
[    0.002769] smp: Bringing up secondary CPUs ...
[    0.003152] Detected PIPT I-cache on CPU1
[    0.003203] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003641] Detected PIPT I-cache on CPU2
[    0.003675] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004166] Detected PIPT I-cache on CPU3
[    0.004199] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004268] smp: Brought up 1 node, 4 CPUs
[    0.004296] SMP: Total of 4 processors activated.
[    0.004310] CPU features: detected: 32-bit EL0 Support
[    0.004323] CPU features: detected: CRC32 instructions
[    0.004374] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004392] CPU: All CPU(s) started at EL2
[    0.004419] alternatives: patching kernel code
[    0.005538] devtmpfs: initialized
[    0.012501] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012542] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.014275] pinctrl core: initialized pinctrl subsystem
[    0.014560] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015163] NET: Registered protocol family 16
[    0.016546] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.017096] cpuidle: using governor ladder
[    0.017126] cpuidle: using governor menu
[    0.017396] Serial: AMBA PL011 UART driver
[    0.018874] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.028050] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-04-15 11:41, variant start
[    0.032047] raspberrypi-firmware soc:firmware: Firmware hash is 1103ec16df2571274600b377411499bf5aed6d0d
[... 811 lines cut ...]
[   79.300069] usb 1-1.1.1.1: no of_node; not parsing pinctrl DT
[   79.303734] cdc_ether 1-1.1.1.1:1.0: no of_node; not parsing pinctrl DT
[   79.305403] cdc_ether 1-1.1.1.1:1.0 eth3: register 'cdc_ether' at usb-0000:01:00.0-1.1.1.1, CDC Ethernet Device, 58:2c:80:13:92:63
[... cut ...]
[ 2589.472765] usb 1-1.1.1.1: New USB device found, idVendor=12d1, idProduct=14db, bcdDevice= 1.02
[ 2589.481501] usb 1-1.1.1.1: New USB device strings: Mfr=3, Product=2, SerialNumber=0
[ 2589.489174] usb 1-1.1.1.1: Product: HUAWEI Mobile
[ 2589.493887] usb 1-1.1.1.1: Manufacturer: HUAWEI Technology
[ 2589.499663] usb 1-1.1.1.1: no of_node; not parsing pinctrl DT
[ 2589.502448] cdc_ether 1-1.1.1.1:1.0: no of_node; not parsing pinctrl DT
[ 2589.504124] cdc_ether 1-1.1.1.1:1.0 eth3: register 'cdc_ether' at usb-0000:01:00.0-1.1.1.1, CDC Ethernet Device, 58:2c:80:13:92:63

Ysurac, 2021-01-22:
You can use "Disable interfaces auto rename" in System->OpenMPTCProuter, "Advanced settings" tab. As interface seems to be renamed and with same dongles this can result in issue.

maksaidd, 2021-01-22:
Hm... Sorry, can't find this option on this tab...

Ysurac, 2021-01-22:
It's in latest release.

maksaidd, 2021-01-22:
Wow! good, I'll update right now!

maksaidd, 2021-01-22:
Done. I set it up one more time. All fresh settings. It worked fine, but after restart

[image]

maksaidd, 2021-01-22:
Then some changes of eth names again and 

[image]

Ysurac, 2021-01-22:
Did you check "Disable interfaces auto rename" ?

maksaidd, 2021-01-22:
Yes, sure

Ysurac, 2021-01-22:
Do you have same issue if you put less dongles ?
What do you change when you change eth ?

maksaidd, 2021-01-22:
1 dongle - OK
even 2 dongles - same issue

I change physical interface for every dongle. 
My plan:
dongle 1 - 192.168.101.1 - eth1
dongle 2 - 192.168.102.1 - eth2
dongle 3 - 192.168.103.1 - eth3
dongle 4 - 192.168.104.1 - eth 4

current situation:
dongle 1 - eth2
dongle 2 - eth1
dongle 3 - eth4
dongle 4 - eth 3

Ysurac, 2021-01-22:
ok, it's what "interfaces auto rename" should fix when enabled. But this doesn't seems to work well...
What do you have as result of ```uci show network``` via SSH or via Services->Terminal ?

maksaidd, 2021-01-22:
That's what I can see on the Terminal page
[image]

maksaidd, 2021-01-22:
Now info from ssh (putty)

network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='5'
network.globals=globals
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.ula_prefix='fd09:6867:be14::/48'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='dc:a6:32:af:04:3b'
network.lan.metric='6'
network.lan.label='Raspberry LAN'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.masterintf='eth0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.macaddr='96:6b:9f:0d:b8:33'
network.wan1.label='BeeLine'
network.wan1.ipaddr='192.168.101.11'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.101.1'
network.wan1.modalias='usb:v12D1p14DCd0102dc02dsc00dp00ic02isc06ip00in00'
network.wan1.product='12d1/14dc/102'
network.wan1.ifname='eth2'
network.wan3=interface
network.wan3.proto='static'
network.wan3.ip4table='wan'
network.wan3.multipath='on'
network.wan3.defaultroute='0'
network.wan3.macaddr='58:2c:80:13:92:63'
network.wan3.metric='7'
network.wan3.peerdns='0'
network.wan3.modalias='usb:v12D1p14DBd0102dc02dsc00dp00ic02isc06ip00in00'
network.wan3.product='12d1/14db/102'
network.wan3.ipv6='0'
network.wan3.label='Ростелеком'
network.wan3.ipaddr='192.168.103.13'
network.wan3.netmask='255.255.255.0'
network.wan3.gateway='192.168.103.1'
network.wan3.ifname='eth4'
network.wan4=interface
network.wan4.proto='static'
network.wan4.ip4table='wan'
network.wan4.multipath='on'
network.wan4.defaultroute='0'
network.wan4.macaddr='58:2c:80:13:92:63'
network.wan4.metric='8'
network.wan4.peerdns='0'
network.wan4.modalias='usb:v12D1p14DBd0102dc02dsc00dp00ic02isc06ip00in00'
network.wan4.product='12d1/14db/102'
network.wan4.ipv6='0'
network.wan4.label='MTS'
network.wan4.ipaddr='192.168.104.14'
network.wan4.netmask='255.255.255.0'
network.wan4.gateway='192.168.104.1'
network.wan4.ifname='eth3'
network.wan2=interface
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.masterintf='eth0'
network.wan2.ipv6='0'
network.wan2.metric='4'
network.wan2.peerdns='0'
network.wan2.macaddr='4e:54:94:3c:61:43'
network.wan2.label='Megafon'
network.wan2.ipaddr='192.168.102.12'
network.wan2.netmask='255.255.255.0'
network.wan2.gateway='192.168.102.1'
network.wan2.modalias='usb:v12D1p14DCd0102dc02dsc00dp00ic02isc06ip00in00'
network.wan2.product='12d1/14dc/102'
network.wan2.ifname='eth1'
network.omrvpn=interface
network.omrvpn.ifname='tun0'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1200'
network.omrvpn.ipv6='0'
network.omrvpn.proto='none'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.255.2'
network.omr6in4.peeraddr='10.255.255.1'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.gateway='fe80::a00:1/126'

Ysurac, 2021-01-23:
ok, same modalias, I will use mac address to recognize each dongle.

maksaidd, 2021-01-23:
OK! Thank you sooo much ;)

pbapba512, 2021-01-29:
I have exactly the same issue with multple Huawei dongle. IMHO mac address can solve this, I'm really waiting for the new build, because I have to setup the whole system after the reboot process. Thank you for the fix, Ysurac!

closed 2021-05-04 by github-actions[bot]

Bestle, 2021-11-24:
@Ysurac Was this ever fixed? I am using multiple dongles and the interface order seems to change every reboot