Ysurac/openmptcprouter #1535: Port Forwarding only works with 80/443 TCP

hrmtc, 2021-01-22:
## Expected Behavior
Being able to port forwarding other ports than 80/443

Being able to forward ports as instructed in https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

## Current Behavior

I'm triying to configure a port forwarding but it only work if the configured port is 443 or 80. 
I don´t see any activity if I run `tcpdump -i tun0 port 8081` on the router side.

I suspect that some firewall rule is blocking other ports but I do not how to check that. 
Could someone please guive me some guidance?

Thanks!

## Specifications

  - OpenMPTCProuter version: 0.57.3 
  - OpenMPTCProuter VPS version: 0.1024 5.4.86 
  - OpenMPTCProuter platform: x86

Adorfer, 2021-01-23:
What you actuall do to provide the forwarding?
My guess is: you are missing at least 1 of 3 neccesary steps.

hrmtc, 2021-01-23:
@Adorfer Thanks for your response.
I found a typo in the WAN VPS firewall.

I'm going to close now this issue.

Thanks for this really useful piece of software!

hrmtc, 2021-01-23:
OpenMPTCrouter was running just fine. The human interface was the problem (sorry).

closed 2021-01-23 by hrmtc